KiCad/kicad-footprints #2: Footprint libs as submodules?

poeschlr, 2017-10-18:
@SchrodingersGat i thought you where against having the footprint libs as submodules.
https://lists.launchpad.net/kicad-developers/msg31020.html

poeschlr, 2017-10-18:
I think i can answer my own question.
One can use submodules to get the history of many repositories merged into one repo.
https://stackoverflow.com/questions/1759587/un-submodule-a-git-submodule

SchrodingersGat, 2017-10-18:
I am against it.. The submodules are all currently existing in `kicad-footprints` as I was performing an experiment... They will not be there for v5. The 'history' will be kept because we will retain the other individual repositories in place, at least for a while.

closed 2017-10-18 by SchrodingersGat