KiCad/kicad-footprints #14: Who is working on what?

poeschlr, 2017-11-08:
I think we should definitely coordinate our effort to avoid doing work twice.

The next thing i will do is look over the JST scripts to move these connectors over here. (Yes i know i did not create most of them but i want to do one complete repo at a time.)
Before that i will transfer the unchanged contents of the Resistors_THT, Capacitors_THT and Diodes_THT repos.

@SchrodingersGat, @jkriege2, @Shackmeister, @evanshultz

SchrodingersGat, 2017-11-08:
I would like to transfer across the `Housings_xxx` repos. However, did we ever decide on a better name for these?

poeschlr, 2017-11-08:
If i remember correctly this came up on the forum. (But i think only as a side node [here](https://forum.kicad.info/t/non-devs-what-do-you-think-of-the-app-names/5147/40?u=rene_poschl))

So i think we never discussed the naming for these libs. (Time to sacrifice a chicken again.)

What about `IC-Package_xxx`

Shackmeister, 2017-11-08:
Ill be working on updating the 3D scripts (mainly naming)
@poeschlr could it make sense to add the height, pin band and pinthickness as parameters in your 2 lead script? this way the yaml could loaded directly in CQ (with some mods to the script) = easier to maintain

poeschlr, 2017-11-08:
Yes this idea came to my mind as well. (But too late.)
If nobody else does it i would do it once all footprints have been transferred. (I would then also rework my connector generators. I would not be mad if someone else does it before i find time for it.)

evanshultz, 2017-11-08:
The discussion on the terms "housing" versus "package" was, I believe, in the old IR-DirectFET repo. But that has been replaced and the porting changed the terms so that's fine.

poeschlr, 2017-11-09:
I decided i will create a generic jst footprint generator. (after the third script where i copied basically the same stuff from my master script.)
So this script will then also have yaml files that define the part dimensions. (I might make it easy for me and make separate scripts for tht and smd versions.)

terjeio, 2017-11-09:
I am currently working on the pin sockets scripts for the legacy footprints, are the ones here script generated with an updated or a new script? If so should I stop my work on and concentrate on the 3D models?

FYI: the legacy script outputs a lot of overlapping and duplicate lines, my updated script reduces the total size by around 25% and, I believe, is more KLC3 compliant.  Also, I have added single row SMD sockects to footprints and 3D models.

Shackmeister, 2017-11-09:
The pinsockets here has only been renamed, you should adapt the Naming to your script, otherwise please continue on with you awesome work :)

evanshultz, 2017-11-09:
So... is this thread simply where we announce what we want to do next? Is there more organization to it? Do we have a list of repos that need moved so we can track progress to completion?

SchrodingersGat, 2017-11-09:
https://github.com/KiCad/kicad-footprints/blob/master/TODO.md

poeschlr, 2017-11-10:
I rewrote the generation script for jst smd connectors. (One script generating all of these connectors.)
This script is also capable of generating other similar components. (Example molex picoblade)
So i think all connector footprints with one row of smd pads plus two mounting pads will be produced by this one script. (As long as the body shape is not too complex.)

Some of the connectors had "inner" details on fab (JST GH). My script is not yet capable of including this. Do you guys think it would be worth my time to add that feature are these inner details not a good idea in general?

SchrodingersGat, 2017-11-10:
Depends on the level of detail you're interested in. I wouldn't go overboard but it may be useful for example to show pin orientation. As long as it shows outline and orientation, those are definitely required.

poeschlr, 2017-11-10:
So show a small square representing each pin for top entry types? (Most side entry types have the pins hidden when viewed from the top.)

SchrodingersGat, 2017-11-10:
That sounds reasonable, would you agree?

jkriege2, 2017-11-10:
Just as a note: I will rework the terminal blocks withing the next few weeks
JAN

evanshultz, 2017-11-10:
I'll take crystals, oscillators, and potentiometers. I want to address https://github.com/KiCad/Crystals.pretty/issues/23 and https://github.com/KiCad/Potentiometers.pretty/issues/18 and https://github.com/KiCad/Potentiometers.pretty/pull/23 and https://github.com/KiCad/Potentiometers.pretty/issues/20.

@SchrodingersGat I recall a PR by you for potentiometers where I had a number of comments and didn't merge. I think they were horizontal Alpha parts, but I can't find an open or closed PR. Do you recall or can you find it?

@jkriege2 Are you going to make shrouded pin headers, both with and without latches? The kinds seen in all 3 pics at http://www.oupiin.com/product_iii.html?c1=10&c2=54? These are extremely common interconnects, and seemingly generic, but I don't see them in our library.

terjeio, 2017-11-10:
@evanshultz 3D-models for some shrouded pin headers are [here](https://github.com/KiCad/kicad-packages3D/tree/master/Connectors_IDC.3dshapes), footprints were [here](https://github.com/KiCad/kicad-packages3D/tree/master/Connectors_IDC.3dshapes) but are deleted?

Since I am working on an update for the pin socket headers footprints and 3D models, which are similar, I can easily add them.

jkriege2, 2017-11-10:
@evanshultz @terjeio Go ahead ... I didn't intend to do the shrouded headers ... although I could if required (but will takes some time!). I'm now starting to redo all those old screw terminals!

Best,
JAN

poeschlr, 2017-11-19:
Above i mentioned that i would look into diodes, resistor and capacitor tht. I fear the connector stuff takes a lot longer than i expected. So if somebody else is motivated in transferring these 3 libs i will not be mad at them.

poeschlr, 2017-11-27:
My next lib will be the old connectors_hirose

antoniovazquezblanco, 2017-11-27:
Hello guys, `Converters_DCDC_ACDC.pretty` has not yet been moved. I wonder if you have talked about splitting it or leaving it as is. Also there is a PR that improves considerably the state of the lib (https://github.com/KiCad/Converters_DCDC_ACDC.pretty/pull/27). 

As I need a footprint from that repo I was thinking in contributing to the switch by opening a PR.

SchrodingersGat, 2017-11-27:
@antoniovazquezblanco I have now merged that PR. If you want to contribute by pulling those across to this repo, that would be greatly appreciated! :) Thanks!

antoniovazquezblanco, 2017-11-28:
Perfect! About naming:
1. Should I split them in ACDC and DCDC or keep them in the same repo?
2. Connector->Conn, then should I rename Converter to Conv?

poeschlr, 2017-11-28:
If there are converters that can work with both AC and DC on the input side then we should keep all converters in one lib. Otherwise it is ok to split them in my mind.
I don't think shortening the name is a good idea here. (I am not even sure any more why we shortened the name for the Conn libs.)

SchrodingersGat, 2017-11-28:
@poeschlr it is the only shortening we do, I'm not sure either. Should we fix all to `Connector_` now, before v5 gets locked in?

antoniovazquezblanco, 2017-11-28:
Converters it is then. Also I will not split the repo yet as it is not that big and also I don't really know if some of the components accepts both AC and DC as input.

SchrodingersGat, 2017-11-28:
\* `Converter_`, singular :)

poeschlr, 2017-11-29:
Next i will transfere the old Samtec and TE-Connectivity repos.

poeschlr, 2017-12-01:
Does anybody know what the stuff in Connectors_Mini-Universal is?

poeschlr, 2017-12-05:
@evanshultz i remember you wrote something about wanting to rework the crystal libs. Does this also include the oscillator lib or is that lib free to be transferred by someone else?

evanshultz, 2017-12-05:
Go ahead. I wanted to fix the issue at https://github.com/KiCad/kicad-footprints/issues/14#issuecomment-343525789, but crystals are already moved. Any change to the fab layers is pretty easy to update later anyway. (The pots had changes to their names which makes it a pain to import and the update later.)

poeschlr, 2017-12-05:
I created an issue to remind us about that particular remaining problem.
https://github.com/KiCad/kicad-footprints/issues/155

poeschlr, 2017-12-06:
I will take care of the Mini-Universal footprints. (They are TE-Connectivity MATE-N-LOK)

SchrodingersGat, 2017-12-06:
Great work! We're almost done :)

Shackmeister, 2017-12-06:
Has most footprint(families) been renamed to match 3.0?

SchrodingersGat, 2017-12-06:
Yes, most. There may still be some that don't match convention requirements but the majority have been renamed.

poeschlr, 2017-12-07:
Next thing i will look at are the remaining footprints in Connectors.pretty
I already transferred the Tag-Connect stuff and am now looking at HE10 (seems to be a locked IDC header, so i will add it to Connector_IDC)

poeschlr, 2017-12-08:
I fear a lot of the footprints in connectors lib are not worth transferring over to the new repo. The only two that might be worth to transfer are the Samtec QMS and TE Future bus. But both would need major rework and checking against the datasheet.
Other then this only the Connector_IEC_DIN and Potentiometer libs remain to be transferred.

evanshultz, 2017-12-08:
Regarding potentiometers, I haven't finished the argparse interface (which seems of dubious value to me anyway...) but all the pot footprints are being generated so I will upload them.

But first, a few questions:
1. Do we want to divide them up in any way? Perhaps into THT (there are ~100) and SMD (~20).
1. I vote for removing the bottom-side ones and providing hole and non-hole versions (the user can flip to the bottom side if they wish). See https://github.com/KiCad/Potentiometers.pretty/issues/18. Objections?

SchrodingersGat, 2017-12-08:
1. Yes, this makes sense to split them at this stage
1. We should not provide duplicates of footprints which are simply "flipped" to the back layer, if that is what you are asking

evanshultz, 2017-12-09:
1. Great.
2. Yes, that is what I was thinking too.

I'll have to make a few script changes to make the second item happen. Probably Mon I can upload.

evanshultz, 2018-01-29:
@poeschlr 
Is there anything to coordinate amongst multiple people? Can this be closed now?

poeschlr, 2018-01-29:
I think the din connectors are still not transferred.

Ratfink, 2018-04-01:
I believe the DIN connectors have now been transferred: https://github.com/KiCad/kicad-footprints/pull/409.

Is there anything left to discuss here, or should this issue be closed?

evanshultz, 2018-04-02:
Close it as far as I'm concerned. I took a close look at the TODO docs in both the symbol and footprint repos. Everything we need should be there now.

closed 2018-04-02 by poeschlr