KiCad/kicad-footprints #46: Testing and travis ideas

antoniovazquezblanco, 2017-11-16:
I had a look to the kicad library utils repo the last weekend. I have some ideas that I am willing to contribute to but I don't know what you think about them.

1. Testing now requires cloning more than one repo (travis clones both this repo and the utils one). I think this could be avoided by moving the testing code to this repo. The reused code between the different testing benches for PCB, SCH and SCHLIB is minimal, thus being a good option. This would reduce check time. Keeping the test code in the same repo would also make easier point number 2.

2. Using relative paths in scripts would ensure that scripts can be run both localy and in travis environment. Currently some scripts can only be used by travis. My proposal is to include a batch and a shell script that the user can click or execute in a terminal that will test the last modified components (I was thinking on testing the whole lib if in the master branch and the modified components if in another branch or something like that).

3. Testing now requires sudo and there seems to be no obvious reason for that. Sudo makes travis testing significantly slower. Dropping sudo needs would improve check times a lot.

4. I was thinking in including a coverage report. A simple script that would tell in which percentage the whole lib complies with KLC and/or another script that would test how many 3D models are present and how many missing footprints looking at the symbol lib.

Please let me know what you think about the previous ideas and if you would like to see them implemented.

Thank you!

SchrodingersGat, 2017-11-16:
1. I don't want to separate out the scripts that are in kicad-library-utils
   a. There is common code that would have to be duplicated
   a. It's a good central place to store the scripts
   a. I think it would clutter the repo which should be reserved for only library files (as far as possible)

2. If you can come up with a scheme that allows this, that's fine, but I don't want to break our current Travis functionality.

3. I didn't know that. That can be dropped from the .travis.yml file I think

4. This is a good idea, I have toyed with this idea before. It would be great to add this output to the https://kicad.github.io website, too! Green indicator for 100% pass, for e.g.

Shackmeister, 2017-11-16:
+1000 for 4.

antoniovazquezblanco, 2017-11-20:
Discarding 1 and 2. Opening issues where it belongs for 3 and 4.

closed 2017-11-20 by antoniovazquezblanco