KiCad/kicad-footprints #74: Some of the second references (on fab layer) for resistor THT footprints interfere with other graphical elements on fab

poeschlr, 2017-11-21:
Some oft he references need to either be scaled down or moved a bit.
Preferably done with the original generator scripts.
But can also be done by hand. (Only a few components are affected.)

grob6000, 2017-12-04:
Looking at this now. Unfortunately http://kicad-pcb.org/libraries/klc/F5.2/ hasn't really considered the axial vertical THT resistor... Plan will be to change the line representing the bent pin to only reach the circumference of the body, and move the R% into the centre of the body. See what it looks like.

grob6000, 2017-12-04:
Having a bit of trouble - keeping this on pause for the moment.
Does anyone (maybe @pointhi, @poeschlr ?) know what the best practice is for the "Fields" in the footprint properties; i.e. "Doc" and "Keywords"? Seeing some differences between what's currently in the footprints and the output of the script and looks like I'm mucking it all up. :)

closed 2017-12-04 by Shackmeister