KiCad/kicad-footprints #78: Diodes THT: Polarity marker on silk not ideal for most footprints and second (fab) reference interferes with other fab graphics

poeschlr, 2017-11-21:
This should be fixed with the generator scripts.

poeschlr, 2017-11-21:
Script location:
https://github.com/pointhi/kicad-footprint-generator

jkriege2, 2017-11-21:
Currently the silkscreen looks like this:
[image]
[image]
[image]

How do you want to change them?

Best,
JAN

poeschlr, 2017-11-21:
Examples where it does not look that nice:

Your last screenshot is a perfect example. If the view is obstructed by other devices next to the diode then it becomes very hard to see the intended direction of the diode. (The assumption is that the polarity mark is used to easily check the correct insertion of the diode when everything is assembled.)
Just imagine a high connector very near to one side and a large capacitor on the other. (Over voltage protection circuit for a DC input)

Other examples (of slightly different problems)
D_5W_P5.08mm_Vertical_KathodeUp (Interference between diode symbol and part outline)
[image]
Even worse if D_DO-201_P5.08mm_Vertical_AnodeUp is placed low down it might be very hard to see the polarity. (or if high components are next to the diode and it is impossible to look from the side.)
[image]

As for the how:
Depends on the exact footprint in question. In your last screenshot maybe elongate The top line to extend beyond the pad? Or you could add a dot, ...
Anything that is visible from directly above.
For others moving the diode symbol might be a good idea. (Where it interferes with other silkline features or is obstructed by the component it self.)

---

On an unrelated note: Do the pads really need that much annular ring? Especially for footprints with small pitch: (might it be a good idea to set a minimum pad to pad clearance within the scripts similar to what i currently do with the connector footprint rework? Would mean if the pitch gets small the pads are not circles but oval.)
[image]

jkriege2, 2017-11-22:
Hi @poeschlr ,
yes I agree that the silkscreen is not that good (I made it that way, when writing the scripts about a year back, as I didn't find any better way ... I think we should assemble some attempts and then select the best one. One idea would be to douplicate the K (for cathode), that we have on F.Fab to make the orientation of vertical diodes clear, on the silkscreen... I think the dot would more easily get lost... Maybe also the offset diode symbol as shown in DO-15_P2.54mm ... any other ideas?

About the angualr rings: YES ... can do ... that's the problem of scripting ... such things just slip through ;-)

Best,
JAN

jkriege2, 2017-11-22:
PS: We should also rethink the  aming scheme (if required) and/or maybe add the specific attributes (`_AnopeUp', ...) to the KLC-listing in http://kicad-pcb.org/libraries/klc/F2.1/

Best,
JAN

jkriege2, 2017-11-22:
PPS: I'm happy to do the scripting, when we decide on a variant ;-)

jkriege2, 2017-11-24:
PR with a fix: https://github.com/KiCad/kicad-footprints/pull/106

poeschlr, 2017-11-26:
I consider this fixed by the merged pull request.

closed 2017-11-26 by poeschlr