KiCad/kicad-footprints #91: Incorrect scaling for THT diode models

SchrodingersGat, 2017-11-23:
Many of the THT diodes have a non-unity scaling for 3D models.

x:y:z must be 1:1:1

SchrodingersGat, 2017-11-23:
This is actually quite wide spread:

* Footprints with incorrect scale - 701
* Footprints with incorrect offset - 339
* Footprints with incorrect rotate - 341

I can change these simply but it will break a lot of 3D models!

jkriege2, 2017-11-23:
Do so ... but we were discussing whether these should be reworked as a hole to improve on the polarity marking on SilkS (see https://github.com/KiCad/kicad-footprints/issues/78 ... maybe you can give your five cents there, so we can move forward) ... When we do that I will also change the scaling (and reexport my 3D models accordingly)

JAN

poeschlr, 2017-11-23:
Wrongly scaled 3d models will not be included in the v5 repo. (Yes i know this will mean we do not have 3d models for some parts when v5 comes out.)

Changing the scaling here would only affect the v5 footprints so i think we can fix the scaling here.

jkriege2, 2017-11-23:
Yes, but it would be good to know WHICH fooprints are affected, otherwise we will never know which models to fix!

So I would vote for doing this in batches: I.e. someone takes a repo, fixes tha scaling and at the same time fixes the 3D models, or removes them!

SchrodingersGat, 2017-11-23:
@jkriege2 yes I think this is a good idea. I'll close my "fix" for this :)

poeschlr, 2017-11-23:
@jkriege2 the models that need fixing are simply the models present in the old kicad-library repo that do not have a counterpart in kicad-packages3d. (Sadly for a lot of them there is no easy fix because they have been designed with wings3d.)

SchrodingersGat, 2017-11-23:
@poeschlr in that case, I will re-open my PR to fix this. It is better to fix the files now and add the models as we go, I think. If there are no existing incorrect models, I think that is fine.

closed 2017-11-24 by SchrodingersGat