KiCad/kicad-footprints #113: LEDs and color

jkriege2, 2017-11-26:
Hi @evanshultz @poeschlr @SchrodingersGat ...

When I recently moved the LED footprints and 3D mdoels to the new repos, I stumbled upon a small problem: All the THT LED 3D models just have any color (most 2-pins red or green, 3/4/...-pin-version are white). That's mostly fine, but there are 3 problems with that:

1. If you want your 3D model to have the correct color, you have to make new 3D models and matching footpritns ... should we provide a list of such? And if so, how do we denote color in the names (e.g. as modifier `_Red`, `_Green`, ...)? 
2. We should have clear/white and maybe grey/black variants at least of the 3mm and 5mm, as they are often used for Phototransistors/-diodes and IR-LEDs ... what do you think? Whould I add them and how should I denote the color in the name (see item 1).

Best,
JAN

SchrodingersGat, 2017-11-26:
You should use the color name as a modifier as you suggest. But which colours?

* Red
* Green
* Blue
* Yellow
* White / Clear
* Black / IR
* ... ?

Also, which is the "default" color (without any modifier suffix)? I would suggest perhaps clear?

poeschlr, 2017-11-26:
For reference, @Shackmeister had an issue in the old repo about LED colors for 3d models:
https://github.com/KiCad/LEDs.pretty/issues/44

SchrodingersGat, 2017-11-26:
Yep I remember @Shackmeister  raising the issue. I'm ok with different colors but not with modelling individual anode and cathodes inside the diode ;)

Shackmeister, 2017-12-04:
@SchrodingersGat SMD led script has now been cleaned for internal parts :P