KiCad/kicad-footprints #117: Nets on wing pads of connectors

Shackmeister, 2017-11-27:
Not sure if this belong here or on the forum, let me know if its inappropriate here.

How are people handling connecting the wing pads on connectors to different nets (typically GND)?
usually I just edit the pads and add the net I want, but this is lost when updating the netlist, which usually makes me forget to change some of the 20ish pads. Unless somebody has a good way, we should maybe consider making a standard padnumber for these. This way the user could use a symbol with or without the wings depending on the nescessity.
Thoughts?

poeschlr, 2017-11-27:
I am currently not understanding what you mean with wing pads. Do you mean the mounting pads like the two larger pads of the picoblade connectors? 
These do not need to be connected to a net because they are just for mounting purposes.

Connectors that have shielding connected to such pins need a pad number of course.

Shackmeister, 2017-11-27:
Yes the larger pads, I typically connect them to GND to have a stronger mount. Sometimes I connect them to other nets to void splitting planes, easier routing etc.

poeschlr, 2017-11-27:
Adding these pin(s) would at least double the amount of generic symbols needed. (And it will add a further complication to users. They now need to answer the question: does my connector come with mounting pads? Even if they don't plan on connecting them.)

poeschlr, 2017-11-28:
@SchrodingersGat @jkriege2 @evanshultz care to comment?

evanshultz, 2017-11-28:
**Oops! I forgot to submit this comment earlier! Sorry.**

In the existing (old) KiCad libs I recall seeing they were universally pin 0 (zero).

There are several situations where a net connection to the mounting tabs is important:
- Mounting strength (mentioned above)
- Shielding (mentioned above, though this may require a specific symbol anyway)
- Chassis ground connection to circuits
- HV / ESD protection
- Connection to touchable surfaces for safety during plug/unplug events

Adding symbols with a pin 0 for mounting pads is good, but I agree that the overhead in the quantity of symbols and footprints that would be added is substantial.

Another option might be to have only one set of symbols, all with pin 0, and the pad pin can be left floating if desired. A corresponding footprint would need a numbered pad 0 which would be floating or connected depending on the symbol's connection. So we eliminate any duplication of symbols or footprints and it's up to the user to make the connection or not.

jkriege2, 2017-11-28:
At least for D-Sub connectors there are symbols with a specific PAD-pin (optional, i.e. variant). the same is done for crystals!

jkriege2, 2017-11-28:
I think at least for those symbols where the mounting/shield-pad is very common, there should be a symbol with it (e.g. RJ-xx, USB, D-SUB, ...)

poeschlr, 2017-11-28:
@evanshultz wrote:
> In the existing (old) KiCad libs I recall seeing they were universally pin 0 (zero).

neither the old Molex, JST, Hirose or TE lib have it like you describe here. (All of them have footprints with either smd or pth mounting pads. None of them have the number 0. Only Molex FFC had a number assigned but it used max pin number +1)

> Another option might be to have only one set of symbols, all with pin 0, and the pad pin can be left floating if desired. A corresponding footprint would need a numbered pad 0 which would be [...]

Not quite. There are a lot of connectors that do not have mounting pins. If the symbols all have the mounting pin in there, the wrong footprints will be suggested by the "by pin count" filter.

@jkriege2 the examples you list are not "only" mounting pins. These are pins intended to be connected because these are shielded connectors that use the mounting pins for connecting the shield to ground. 
They of course should have a pin number.

With mounting pins i mean explicitly the two pins at the outside of for example picoblade smd version. These are small metal parts on the side of the connector encased in plastic.

poeschlr, 2017-11-30:
I now have a connector with shielding. This should get a pin number. We currently don't have a symbol for it. I would generate one. But what pin number to give?

I might even vote for "SH" to make it really clear what this pin is. (Similar to the use of "EP")

poeschlr, 2017-12-05:
I added the symbols in https://github.com/KiCad/kicad-symbols/pull/26. There is used SH as the pin number.

Shackmeister, 2018-01-30:
if we now have symbols with SH could we possibly use this for all the mounting pins?
I would really love this to get implemented

poeschlr, 2018-01-30:
Mounting pins are not shields!

poeschlr, 2018-01-30:
The only option i see here is to add another set of generic connector footprints that have a mounting pin. And hope that there is never a connector that has mounting pins and shield that is not connected.

If we go this route at least the single row connectors must specify the pin count in their footprint filters. (Maybe they also should do it now, as the shielded connectors might be suggested for non shielded symbols with pin count +1)

Shackmeister, 2018-01-30:
Hmm thats true, so a seperate symbol with MNT perhaps?
and agree the pincount must be specified in the FP filter

poeschlr, 2018-01-30:
We would also need a suffix in the footprint names for either only the mounting pin or for both mounting pin and shield pin. (or we would need a similar syntax for the pin number as we have for exposed pads.)

poeschlr, 2018-01-30:
I started a poll over at the forum to get the opinion of a wider audience: https://forum.kicad.info/t/official-library-poll-should-mounting-pins-for-connectors-have-a-connection-on-the-schematic/9483

poeschlr, 2018-01-31:
@jkriege2 and @evanshultz i would like your input here before i make my final decision.

---

The forum shows there are some who would like the possibility but also some who think the price to pay is too height. (I count my self to the later.)

I might allow it if all the following conditions are met:

- Only one pin in the symbol (all mounting pins get the same pin number)
- The pin number for the mounting pad is not a number. (Example: MP or MTP)
- The symbols are in a separate lib (Connector_Generic_MountingPin)
- **All footprints** are updated. (If a script exist this has to be done in the script)
- Both symbol and footprint changes are done in a timely manner before the version 5 release.
- The mounting pin footprints get a suffix _MountingPin (Or should we add -1MP after the pin count in the footprint name. This might be more flexible if we ever want to allow multiple mounting pins with differing pin numbers)

I will make my decision within the next few days.

jkriege2, 2018-01-31:
Hmmm ... I don't really see the difference between shield and mounting pins (for any practical purpose) ... I would see it that way: 
1. If there are mounting-holes/pads that are connected to some kind of metal-coated housing: I would assign them a specific pin-number that allows to use it as a connection for the shield (all D-SUB use pin-number 0!!! USB are a mix of No. 5, 6, S1, HDMI uses 20 ... so it's a real mess) ... for sure we need to define that in KLC ("shield" does not appear there)
2. If they are for mounting only and no shielding can be expected from them (What would that be? A plastic connector with glued-in/molten-in pins???), I don't see the point of connecting anything to them. They are only there for mechanical stability
3. Are there any packages that have both? If so the pins for shields should follow rule 1 and the rest rule 2

Can you give examples?

Anyways: If we come to any conclusion we need to make sure to add it to the KLC and unify the lib to match to that rule!

poeschlr, 2018-01-31:
For now we are talking explicitly about these kind of pins (your option 2)
[image]

They only have one purpose: Making a mechanical connection between the connector housing and the pcb. There is no shielding effect from them. There is no electrical contact to anything other than the pin it self.

---

I don't talk about shield pins that have a dual purpose. Yes such pins also make a mechanical connection to the pcb. But from a pure electronic point of view their purpose is to connect the shield. (We already have generic connectors for such shield pins. There is the Connector_Generic_Shielded lib)

---

I am sure there are also connectors that might fall into category 3. If we also want to include symbols that offer different connections for their shield and mounting pins, the number of symbols will really start to explode. For now i would definitely not include them. (We don't even have footprints that would fall into that category right now.)

jkriege2, 2018-02-01:
Hmmm ... I still don't see the real reason for connecting them to GND (or some other signal)? Is the idea that if I solder them onto a ground PLANE (i.e. larger peace of copper), the holding-forces are stronger, because any pull is distributed over a larger interface between PCB and copper???

antoniovazquezblanco, 2018-02-01:
Soldering them to a plane gives much more mechanical resistence. Also, when tightly fitting this kind of connectors in the border of a board, those pads can help stich ground planes eliminating the need to create vias.

Shackmeister, 2018-02-01:
stuff like this can typically be reduced quite alot by connecting the mounting pads to a solid plane.
and as mentioned before it can also be useful to avoid splitting planes
[image]

poeschlr, 2018-02-01:
I doubt a plane will help here. Copper is too weak to add much support. I think the only thing that will help you are vias inside the mounting pad. Not sure if one can add vias without a net though.

Shackmeister, 2018-02-01:
compared to the strength of the bonding for the pad the plane is definitely stronger.
I just tested a few boards, havent been able to break the pad. Solder or connector breaks before the pad.
but yes via's are also a good thing to add for strength

evanshultz, 2018-02-01:
As I mentioned above, there are various reasons for connectors to have these pins. In my opinion, the option to have a zone flow solidly under the connector's mounting pins is reason enough. These are cases where board design rules would require an unbroken zone, especially around a board edge.

I suggest only 2-character names (SH, MT, etc.). That works with 100mil pin length and 50mil text size.

-1MP or something similar is a good name for the footprints for me. It also allows a nice FPfilter.

Causing ERC errors or design rule violations because of the symbol/footprint design seems poor. I'm not sure of the best pin type, but I do not think we should inherently cause user issues due to the symbol or footprint library's design.

I understand the entire discussion is only about generic connector symbols for these types of footprints. Symbols which are for a specific connector should have the appropriate pins, of course, and should match the pin naming convention for consistency. So if any symbol pin is a shield, it will be `SH`. If any pin is only for mounting, it is `MT`. If it is only for ground, then the pin name is `G`. And so on. (I made up those pin names but something like that should be codified in KLC once this is settled.)

poeschlr, 2018-02-01:
@Shackmeister as promised now my decision.

You can go ahead under the requirements i already listed above.
Regarding pin type i fear our only option is to use passive. (Only alternative NC will result in ERC warnings for people who wish to connect them. That ERC warning can not be turned of for that specific pin alone. Whereas users who do not wish to connect a passive pin have the NC flag to tell ERC that this is on purpose.)

Regarding footprint name: I would go with -1MP.
Pin name should be MP.
I don't think there will be a connector with a ground only pin that is not a shield. So i guess there will always only be either MP or SH pins. We will not prepare the lib for connectors that have both MP and SH pins.

As stated in another issue, i would also vote to add -1SH to footprint names for shielded footprints. (But this is a separate issue.)
This change will only make it into v5 if you get it done for all footprints until the tag for RC1 should be created. (I would prefer if we had a bit of time left. Such that we can fix remaining issues beforehand.)

Shackmeister, 2018-03-10:
@poeschlr should all the affected connectors get -1MP added to their name?

- Molex_PicoBlade_53398-0971_1x09_P1.25mm_Vertical-1MP

- Molex_PicoBlade_53398-0971_1x09-1MP_P1.25mm_Vertical

Shackmeister, 2018-04-04:
As these has been merged Im now closing this issue, thanks for all the work

closed 2018-04-04 by Shackmeister