KiCad/kicad-footprints #130: What about manufacturer specific pin headers and socket strips? (Samtec)

poeschlr, 2017-11-28:
Currently the samtec lib holds some pin header footprints. Should we transfer these specialized footprints over here? (They have slightly different drill sizes: 0.95mm compared to 1mm for the generic footprints)
Samtec creates quite a lot of slightly differing pin headers and socket strips. I am not sure it would be a good idea to include specialized footprints for all of them. (But i am also hesitant to simply "delete" contributions.)

evanshultz, 2017-11-28:
IMO, these are generic enough that we can use a "one size fits all" and eliminate vendor-specific footprints where possible. This should result in the least maintenance issues for librarians while still supporting all components a KiCad user would want to use. I share the guilt from removing someone's hard work from the library, but to me it seems the best way to proceed.

terjeio, 2017-11-30:
I have just remade the script for the pin sockets and when I did the research for them I found that dimensions can vary quite a bit between the vendors. My plan is to dig into this issue in more detail to see which variants is not covered by the existing ones and possibly add them. Bottom entry, guide pegs, different land patterns...

3D models is another issue, varying heigths, turned pins...

How much work should be put into this? I guess it depends on the level of accuracy and coverage the librarians are aiming for.
Also, if I am going to do this I will need some input on when a new footprint should be made. Eg. I have a SMD pin socket on my desk that requires a wider land pattern than the current KiCad version provides.