KiCad/kicad-footprints #133: SIP3_11.6x8.5mm is not compatible to RECOM R-78

mistoll, 2017-11-29:
AFAIK the footprint kicad-footprints/Package_SIP.pretty/SIP3_11.6x8.5mm.kicad_mod is the replacement of Converters_DCDC ... RECOM R-78 footprints.
Both, the original footprints in the Converters_DCDC package and the SIP3_11.6x8.5mm are not compatible with the RECOM R-78 parts.

What's wrong:
The courtyard and silk have the large part upwards, while is should be downwards. The datasheet [R-78Bxx-1.5](https://www.recom-power.com/pdf/Innoline/R-78Bxx-1.5_L.pdf) has a recommended footpring graphik on the last page, with pin 1 marked.

poeschlr, 2017-11-29:
Is this really the replacement for only these converters? @evanshultz i think you looked at the sip lib a while ago.

If so the footprint should be moved anyways. (All other converter footprints will be in the Converter lib added by https://github.com/KiCad/kicad-footprints/pull/127)

evanshultz, 2017-12-05:
Thanks for looping me in. I'll take a look tomorrow.

evanshultz, 2017-12-05:
Neither https://github.com/KiCad/Converters_DCDC_ACDC.pretty nor https://github.com/KiCad/kicad-footprints/tree/master/Converter_DCDC.pretty have a R-78Bxx-1.5 footprint. It would appear to have never existed. The new repo does have 6 RecomR-78 footprints, so I think the proper footprint should be created, if desired.

The SIP-3 is totally unrelated to any Recom parts as far as I can see.

@mistoll Why would you think there was a footprint replacement? It would be welcome if you could make the missing footprint.

@poeschlr I think this case can be closed unless some interesting information develops.

jkriege2, 2017-12-05:
Hi!

the original DCDCs from ReCOM are script-generated: https://github.com/pointhi/kicad-footprint-generator/pull/70

ideally, please modify the script to get a consistent style!

Best,
JAN

mistoll, 2017-12-06:
@evanshultz You are right, there's no R-78Bxx-1.5 footprint. 

I referenced that datasheet, because the footprint layout graphik is unambigous there. While in other R-78 datasheets the footprint layout doesn't mark pin 1, thus allowing failures.

If the package SIP3_11.6x8.5mm is unrelated the line 
https://github.com/KiCad/kicad-footprints/blob/cf6453685279d9cad03ca35fd590b9d91e7af775/Package_SIP.pretty/SIP3_11.6x8.5mm.kicad_mod#L2 should be removed/changed.

Sorry, I currently have not enough time to fix those footprints, especially because I have no experience in sciprt generating footprints.

Werni2A, 2020-07-29:
@evanshultz I think there is some kind of misunderstanding. `Package_SIP:SIP3_11.6x8.5mm` is meant for [R-78E-0.5](https://www.recom-power.com/pdf/Innoline/R-78Exx-0.5.pdf) and is therefore unrelated to [R-78B-1.5](https://www.recom-power.com/pdf/Innoline/R-78Bxx-1.5_L.pdf) which requires a different footprint.

However the script generated `Converter_DCDC:Converter_DCDC_RECOM_R-78E-0.5_THT` can replace `Package_SIP:SIP3_11.6x8.5mm` as they are the same footprint. I'm not sure how to check that no symbol in the lib uses this footprint and would break if I remove it. Is there some script to verify this? I tried to search for the footprint in the symbol library which returned no results.

If you want I can create a PR to remove `SIP3_11.6x8.5mm`