KiCad/kicad-footprints #144: Button_Switch_SMD.pretty naming convention not continuous

pointhi, 2017-11-30:
We have some mixture of rotary encoder without a prefix, switches without a ```SW_SPST``` prefix and switches with/without manufactur name

Ratfink, 2018-04-01:
Also: a lot of pushbutton switches are currently prefixed with `SW_SPST`.  Should they be renamed to use an `SW_Push` prefix?

Shackmeister, 2018-04-07:
SW_Push doesn't give the electrical configuration of the switch. A pushbutton could also be SPDT, SPDT etc

Ratfink, 2018-04-07:
Good point.  I think it would be best to put both in the name, which would allow for better footprint filters.  Really, our switch naming needs a lot of improvement and standardization, and it should extend to the symbols as well.

poeschlr, 2018-08-29:
We can think about a scheme that is used for new contributions. But existing symbols and footprints should not be changed in a manner that would break existing projects that use them. (Otherwise we are unable to supply users with bugfix updates of the library.)

poeschlr, 2019-01-16:
We have documented a naming convention for new contributions here: https://github.com/KiCad/kicad-symbols/issues/580