KiCad/kicad-footprints #155: Polarity markers for Crystal footprints is missing

poeschlr, 2017-12-05:
For details refer the original issue over at the old repo:
https://github.com/KiCad/Crystals.pretty/issues/23

---

It seems most of the footprints are scripted. So fixing this inside the scripts is preferred:
https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Crystals_Resonators_SMD/make_crystal_smd.py
https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Crystals_Resonators_THT/make_crystal.py

evanshultz, 2018-01-29:
From #296:
Polarity markings aren't in accordance with IPC: https://github.com/KiCad/Crystals.pretty/issues/23 and https://github.com/KiCad/Oscillators.pretty/issues/6. Note that most (all?) are script-generated with more info at the first link.

https://github.com/KiCad/Crystals.pretty/issues/24 mentions a possible seems to point out a mismatch between a package name and the package size.

evanshultz, 2019-01-06:
I'll avoid changing the title of this issue for continuity reasons, but the zero orientation of the crystal footprints should be addressed at the same time when the script is updated/re-written.

Actually, I'm rather interested in doing this once I get one more big item off my plate.