KiCad/kicad-footprints #158: What should we do with the wire pads footprint lib?

poeschlr, 2017-12-05:
The test point lib has been transferred in https://github.com/KiCad/kicad-footprints/pull/52
The question about the wire pads lib has been raised in that pull request already.

SchrodingersGat, 2017-12-05:
I think that wire pads are functionally separate from e.g. test points, so we should add a new library e.g. `WirePad`

poeschlr, 2017-12-05:
I found an old discussion where you suggested the name Connector_Wire. (Would make more sense.)
https://github.com/KiCad/Wire_Pads.pretty/issues/5

I think i will for now use that name for it.

---

In that issue the deprecated lib [Wire Connection Bridges](https://github.com/KiCad/Wire_Connections_Bridges.pretty) is mentioned.
From the discussion i would amuse you meant to give it the name Connector_Jumper? (But under jumper i would understand this: https://i.stack.imgur.com/UsiJ3.jpg not https://circuitmaker.com/forum/posts/216769)
I think in that case `Wire_Bridge` would be better. In addition i don't think we currently have any true wire bridge footprints in our lib.
I guess a better idea would be for users to use vias and unused copper layer(s) to document the connections. (This would allow for doing this without needing to add a symbol to the schematic.)

In any case i think that particular discussion can wait until somebody does indeed contribute such a footprint.

SchrodingersGat, 2017-12-05:
The way I see it, `Connector_` implies that some external "thing" will connect to the PCB. I think it also implies that this "thing" is removable, perhaps?

Jumpers do not fit this category IMO. By Jumpers I mean:

[image]

[image]

[image]

[image]

[image]

evanshultz, 2017-12-05:
To me, "wire pad" means a THT (very rarely a SMD) pad to which a wire is soldered. There is no other discrete component, just the wire.

In my vocabulary, here are the names of the items in the pictures above from top to bottom:
- Testpoint or testpad
- Jumper (this does not have a symbol or footprint since it fits on a header)
- Solder short
- Either a soldered jumper/bridge or a high-current, low-impedance sense resistor
- Either another soldered jumper/bridge or a high-current, low-impedance sense resistor

I agree that wire pads are a separate category and different from any of the pictures above. If I see a library named `WirePad` I would expect to see plated THTs named by either mm or AWG for fitting one or more wires.

myfreescalewebpage, 2019-03-15:
@poeschlr @SchrodingersGat @evanshultz WirePad I have seen Connector_Wire has been created following this issue, do you still need to keep it opened ? Else I suggest the topic should be closed. Joel

evanshultz, 2019-03-15:
I'm fine to close this then. However, I notice at least some (I didn't check all) seem to have converted dimensions because the names are "clean" mm sizes but the actual pad sizes have loads of significant digits.

myfreescalewebpage, 2019-03-15:
Great catch, maybe something we can fix with a PR, I know several footprint have this problem... Some improvements of the travis checks for F5.3 (http://kicad-pcb.org/libraries/klc/F5.3/) should be great to check that:

> All courtyard line elements are placed on a 0.01mm grid

closed 2019-03-15 by myfreescalewebpage