KiCad/kicad-footprints #160: What should we do with the microwave library

poeschlr, 2017-12-05:
I am not really sure what these footprints represent.

Some of them have smd pads with all copper layers set
Not sure they should include paste.

poeschlr, 2017-12-05:
It seems they suggest to drop this lib.
https://forum.kicad.info/t/another-try-does-anybody-know-something-about-the-microwave-footprint-repo/8766

jkriege2, 2017-12-05:
I would also say so ... even if they are tuned for something (see e.g. https://www.jlab.org/accel/eecad/pdf/050rfdesign.pdf for PCB designs for antennas and other RF-stuff), they are worthless without the metainfo.

SchrodingersGat, 2017-12-05:
Yep, let's drop it. If anyone ever complains we can add it back in. I doubt anyone will.

poeschlr, 2017-12-06:
Added the info that this will be dropped to the todo file

closed 2017-12-06 by poeschlr