KiCad/kicad-footprints #171: Should THT components include a courtyard area on the bottom?

poeschlr, 2017-12-08:
I think kicad now has a DRC check for courtyard violations.
But our THT footprints don't include courtyard on the bottom. So if one would place something too near on the bottom then this check would not notice it.

Shackmeister, 2017-12-08:
Can you provide some image examples?

evanshultz, 2017-12-08:
What is the problem you would want to catch? I've never ran into issues _only_ with placing parts under a THT component.

There can be issues with SMD parts on the bottom placed too close to THT pads, but the exact requirements depend on if the board uses double-reflow or wave for bottom SMT, if the wave has solder jets, if a selective solder pallet is used, etc. This depends greatly on the manufacturing equipment, process, and operator skills. And the actual constraint to be controlled is bottom THT copper pads to bottom SMD copper pads, anyway. Nothing about courtyard I've seen.

A THT on the bottom under a THT on the top could have lead interference. Is that it? If that's the issue you're thinking about, can KiCad discriminate between a THT part and a SMD part on the bottom (SMD wouldn't have leads poking out the top of the board).

poeschlr, 2018-02-02:
It seems the answer is no.

closed 2018-02-02 by poeschlr