KiCad/kicad-footprints #174: wrong sizes for R_1020

Shackmeister, 2017-12-09:
@poeschlr 
Hi Rene I noticed R_1020 and the handsoldering version both has a body size of 5x2.5mm, this seems quite big :)
can you look into it?

[image]
[image]

poeschlr, 2017-12-10:
Size is ok. But metric code is wrong.
[image]

poeschlr, 2017-12-10:
should be fixed now. @Shackmeister close this if you are satisfied.

Shackmeister, 2017-12-10:
Sounds good, ill check it out tomorrow

poeschlr, 2017-12-16:
@Shackmeister is this fixed to your satisfaction?

Shackmeister, 2018-01-03:
Hi @poeschlr this is only fixed for the Hansoldering version, not the regular

Shackmeister, 2018-01-07:
@poeschlr will you have time to fix this or should I do it? :)

closed 2018-01-09 by jkriege2