KiCad/kicad-footprints #188: Issues with potentiometer footprints

evanshultz, 2017-12-21:
This will be a living list as the footprints are finished up and merged. Some of them will have the script as the root cause and re-writing or extending the script wasn't in scope for v5.

General:
- Pin number ordering (CW/CCW/etc.) cannot be specified. This prevents [Bourns PRS11S](http://www.bourns.com/docs/Product-Datasheets/PRS11S.pdf) from being built to the datasheet.
- Pinxoffset variable controls body placement as well. (Pins, body, and shaft should be independent.) This prevents  from being generated correctly.

SMD:
- SMD pad sizes are all the same. [ACP CA6 VSMD](http://www.acptechnologies.com/wp-content/uploads/2017/06/01-ACP-CA6.pdf has two different sizes, for example.

poeschlr, 2017-12-23:
About pin ordering: We have some footprints in the LGA lib that already add the suffix ClockwisePinNumbering to it. (No suffix = standard counter clockwise)

SchrodingersGat, 2018-01-06:
@evanshultz how are the potentiometers coming along? This is the major remaining item for this repo and we need to be able to tag a library release very soon...

evanshultz, 2018-01-12:
@SchrodingersGat 
They were transferred some time ago. See https://github.com/KiCad/kicad-footprints/pull/190. It seems I forgot to update the TODO doc in that PR.

@poeschlr 
Clockwise or counterclockwise doesn't apply to the pin order on footprints. I'll write up a new issue to capture the missing link because the schematic symbol and the pin numbers.

@jkriege2 
On a personal note, I feel I owe you all an explanation and an apology. About 5 weeks ago I suddenly started feeling very ill and was diagnosed with a brain tumor. It greatly affected my ability to think and function and I'm sorry that I was unable to contribute during this critical time as KiCad approaches v5.

Fortunately, with faith and a great medical team the tumor is removed and I'm feeling better now. I'll be able to start contributing again while I continue to recover.

I apologize again for disappearing. While I have a good excuse, there are so many GH emails it's clear another hand would have been helpful. If you tag me with a request now I should be able to help!

SchrodingersGat, 2018-01-12:
Ok, I've ticked that off the TODO then. Thanks

evanshultz, 2018-02-06:
@poeschlr @jkriege2 
When reviewing https://github.com/KiCad/kicad-footprints/pull/349, I noticed that the script doesn't generate a square pad for pin 1 of any pot footprints:
[image]

KLC requires this, and some (maybe all?) of the footprints are directional and can only be inserted one way so I don't think this is critical to fix immediately, but we use square pin 1 pads on other THT footprints that are directional too. Unless someone tells me different, I'll add this as an issue.

jkriege2, 2018-02-06:
I think pin1-indicator is only required for polarized devices, like diodes or electrolytic caps. We also don't use them on normal resistors or caps ... for pots they are (IMHO) only necessary, if we go through with the CW/CCW-markings ... and even then maybe another scheme of marking makes more sense?

poeschlr, 2018-02-07:
Marking one of the pins might still be helpful for debugging the pcb.