KiCad/kicad-footprints #191: Missing audio connectors

evanshultz, 2017-12-22:
https://github.com/KiCad/kicad-footprints/blob/master/TODO.md doesn't include (or doesn't include many) various professional audio connectors:
- 1/4" / phono
- XLR
- speakON

Some of these footprints exist in the old library but are outdated. I'll update and expand this collection of audio-specific connectors.

poeschlr, 2017-12-22:
Nice catch

terjeio, 2017-12-27:
@evanshultz I want to make 3D-models for Pro-Signal PSG01550 and PSG01551 RCA/phono sockets, I have footprints for them already but most likely not KLC compliant so needs to be updated. Are these in the list of those you want to add or should I keep them on my todo list?

evanshultz, 2018-01-12:
Go ahead. I'm going to be adding only Neutrik and Amphenol pro audio connectors (XLR, speakON, etc.) to start because they're generated considered "standard" or "reference" and most other brands of connectors are compatible. Thanks for the proposed contribution!

evanshultz, 2018-01-18:
@poeschlr @jkriege2 
Question on pin numbers for these audio connectors that don't have pin numbers that are sequential integers starting at 1. Here are some examples (see the PDF drawing):
- http://www.neutrik.com/en/speakon/speakon-chassis-connectors/nl8md-v
- http://www.neutrik.com/en/audio/plugs-and-jacks/1/4-vertical-jacks/nj6fd-v
- http://www.neutrik.com/en/audio/plugs-and-jacks/jacks/combo-a-series/ncj6fa-v

Shall we give them normal audio conventions (and have a corresponding symbol) which I'm used to? That would mean the following pin numbers (in order from the list above):
- 1+, 1-, 2+, 2-, etc.
- T, T+N, R, R+N, etc.
- 1, 2, 3, T, R, S

If we use this pin numbering, where should the symbol be centered? Most of them will not have a pin numbered simply `1`. Do we use the closest pin number to 1? Or center the body geometry?

jkriege2, 2018-01-19:
For the (switching) audio-jacks, there exist generic symbols, e.g.
[image]
So if possible I would make the footprints compatible to that symbol. 

For the XLR- there are also generic symbols XLR3-XLR5, so I would make the compatible to thise, e.g.
[image]

For the combines XLR+audio-jack, I would define a new symbol that is specialized.

One could think about adding a second variety of the symbols with some kind of postfix like `_StandardStereo` or something and then mention in there as text the names of the standard audio-signals, maybe as pin-labels opposite to the pin-number (is there a global standard for those or is it just everybody does it that way to be compatible? In the first case, we should also name the standard).

For the footprint centering: I think here we should for now stick to the THT:pin1=(0,0), SMD:pkg_cntr=(0,0) rules, but maybe discuss whether we modify them for switches and connectors, e.g. like this:
1. vertical switch/connector: (0,0) marks the connector/switch main axis/fronpanel dril-hole-center ...
2. hor. switch/connector: (0,0) is on the switch/connector-axis and maybe on the proposed PCB-edge

I think that would also work for pots ... or generally for everything that can be operated by hand. We would get problems with the check scripts though and I'm not sure about my rule 2 above!

What do you think?

JAN

evanshultz, 2018-01-19:
XLR uses just pin numbers, and that's fine with me.

But switching jacks have pin numbers that aren't standardized, in my experience, so using intelligent pin numbers seems best to me. It's far easier to check a schematic and PCB if the pin numbers are helpful rather than just numbers. For example, the symbol you showed above uses 1, 2, and 3 for S, R, T (normal audio convention would be T, R, and then S). And the 4-6 for the switched outputs. Using T, T+N, R, etc. is much more obvious and I believe would help KiCad users make sure their design is built right.

Since Neutrik is so well-known (and copied!) for audio connectors I put significant weight on using their naming conventions.

But, it would mean non-number pin numbers. We've talked about this a few times (I recall a discussion regarding switches but can't find it now). I'll rework the audio connector symbols (and footprints, if needed), and add new ones, if this is OK with y'all.

jkriege2, 2018-01-19:
What do T,R,S,T+N,... stand for?
Then why not use 1,2,3,.... for the different rings on an audio jack, starting from the tip, numbering towards the mantling and then using S1, S2, S3, ... for the switches where the number indicates the ring they are connected to ... wouldn't that make sense + it allows for extendability of the system to any such connector.

jkriege2, 2018-01-19:
Also I noticed Neutric using RM and RN contacts .... but I'm not so clear what these names stand for ...

evanshultz, 2018-01-19:
@jkriege2 
It sounds like you're looking at a combo XLR + 1/4" connector like NCJ10FI-V? The poles of an XLR are numbered 1, 2, and 3, with well-defined connections (1 = ground, 2 = +, 3 = -). Having switched terminals is commonly called "normalling", and thus the nomenclature of "N" terminals. The poles of a 1/4" stereo are also well-defined but use a different terminology (T = tip, R = ring, S = sleeve).

And XLR connector may use 1-3. A stereo 1/4" may use T, R, and S. But a combo connector, as described above, would naturally use both since the terminals are distinct and the signals to/from the connector are implementation-dependent so they cannot be known. (For example, the XLR and 1/4" may go to different circuits. Or they could be wired together. This is only known by the designer.)

Where did you see RM contacts?

speakON connectors (like NL4MD-V) have two poles for each terminal, which could be grounded or not, and adopt 1+, 1-, 2+, etc. markings. Regardless of the electrical connection, one signal (1-) will be the reference for the other (1+) so they go together. Also, this marking indicates polarity/phase through the device (audio input to audio output) which is also extremely important in an audio system.

If we don't adhere to this well-understood terminology in the audio industry for KiCad's audio connectors, then I believe (with relevant experience) our symbols and footprints will be unintuitive.

jkriege2, 2018-01-19:
I found that there: http://www.neutrik.com/zoolu-website/media/download/2804/Jack+Circuits (lower row)

evanshultz, 2018-01-19:
If the jack is normalling (typically passes through, but is broken when a terminal is plugged in), the nomenclature is *N.

I'm honestly not sure about the "sleeve make" and "ring make" versions, but I suspect they're the opposite of a normalling jack in that plugging in a terminal makes a connection instead of breaking it. This might be used for an effects send/return connector. The nomenclature for those signals is *M while the other terminals are normalling with the common nomenclature.

Does that answer it?

evanshultz, 2018-01-22:
@jkriege2 @poeschlr 
### 1. Audio connector pin names
Any other feedback on the above?

I have several questions after making a good number of footprints and thinking about this.

### 2. Footprint names
I think http://kicad-pcb.org/libraries/klc/F3.6/ covers these audio connectors in `Manufacturer Specific Connectors` section (many alternates are copies of a well-known vendor and can be linked back to the original vendor). If I read it right, here are some valid examples of correct footprint names that I'd like feedback on:
- [`Neutrik_XLR-AA_NC3FAAV_Vertical`](http://www.neutrik.com/en/xlr/aa-series/nc3faav)
- [`Neutrik_XLR-AA_NC3MAAH_Horizontal`](http://www.neutrik.com/en/xlr/aa-series/nc3maah)
- [`Neutrik_speakON_NL4MD-V_Vertical`](http://www.neutrik.com/en/speakon/speakon-chassis-connectors/nl4md-v)
- [`Neutrik_XLR-6.3mm_NCJ10FI-V_Vertical_Combo`](http://www.neutrik.com/en/audio/plugs-and-jacks/jacks/combo-i-series/ncj10fi-v)
- [`Neutrik_6.3mm_NMJ6HCD2_Horizontal_Stereo_Switched`](http://www.neutrik.com/en/audio/plugs-and-jacks/m-series/nmj6hcd2) (this is TRS stereo, and `Stereo` would be omitted for a mono TS version)

### 3. Existing footprints
I took a look at the existing footprints in :
[image]

They all need some work:
- `Jack_Horizontal` has a description in French and no identifying info
- All others but the first one appear to have a PN but no datasheet or manufacturer
- `Neutrik` is misspelled on the first one

Proposed fixes:
- I can correct the spelling when I submit my PR.
- But what about the unidentified connectors? Should we remove them if we can't offer a KiCad user what component they need to fit the footprint?

evanshultz, 2018-01-22:
### 4. Fab vs silk drawings
On a part like [`Neutrik_6.3mm_NMJ6HCD2_Horizontal_Stereo_Switched`](http://www.neutrik.com/en/audio/plugs-and-jacks/m-series/nmj6hcd2), which sits flush against the PCB in some areas but also hars "ears" for mounting above the PCB, how do we indicate this? I think we do fab lines everywhere, and silk in both areas. This lets the user see the connector placement against the PCB and also shows the orientation of the connector by putting the ears in silk. Here is an example:
[image]

I put fab lines everywhere but only silk around the outer extents. I think it's best if I also add silk where the connector touches the PCB. Any thoughts on that?

poeschlr, 2018-01-23:
Do not loose to much time on unidentified connectors. There is a reason we did not transfer all connectors from the old generic connectors lib.
If they are already in the new lib, please remove them.
About silk: looks good to me as you made it.

poeschlr, 2018-01-23:
About footprint name: I think they are not comparable to the manufacturer specific connectors but more like sub-d or usb connectors. This is called connectors for a specific function. (old name: standardized connector interface but manufacturer specific landing patter)

There the naming convention is:
`[Function]_[Standard]_[Size]-[Type]_[MAN]_[MPN]_[Modifiers]_[Orientation]_[Options]`

Example from your list:
`XLR-AA_Neutrik_NC3FAAV_Vertical`

jkriege2, 2018-01-23:
If the connectors are meant to work with some geneirc symbols (like the jack-symbol-series), we should also take into account whether it is possible to have a name-part that allows to filter them auto (at least to some degree of reliability) with the symbol FPFPilters ... I think the StereoJack_3.5mm-prefix is one way to do that.

evanshultz, 2018-02-05:
@poeschlr 
What about `XLR_Neutrik_NC3FAAV_Vertical`? The "AA" part is the series of Neutrik connectors, and that's already encoded in the MPN.

And regarding pin numbers, what are we doing? I propose to use Neutrik PNs where `T` is the tip of a 1/4" jack, `TN` is the normalling tip pin, etc. Some of the XLR jacks have a dedicated ground pin, where we've discussed recently calling `SH` or `MP` on other connectors. Take [NC3FAH](http://www.neutrik.com/en/xlr/a-series/nc3fah), [NC3FAH1](http://www.neutrik.com/en/xlr/a-series/nc3fah1), and [NC3FAH2](http://www.neutrik.com/en/xlr/a-series/nc3fah2). They are all XLRs but with different grounding schemes. The first one just has the 3 XLR pins so they would be pin 1-3, the middle one connects one XLR mounting screw (shield of the cable if the cable is shielded) and chassis to pin 1 but still keeps only 3 pins, and the last one has 4 pins by adding a dedicated ground. What should these pins be named? The first one is easy, but what about the last two? Does the middle keep just pins 1-3 and leave it up to the user to realize that this connector also shorts pin 1 to the XLR mounting screw? And what does the last one name the ground pin? I personally don't think `SH` is the best name for the 3rd connector's ground pin because this isn't guaranteed to be a shield, so perhaps `G`?

evanshultz, 2018-02-05:
Here is an example of a company making largely compatible XLRs to Neutrik: http://www.zgele.com/products/XLR-CANNON-COMBO-CONNECTOR-SERIES.html. This includes the pin number of http://www.zgele.com/product/High-quality-Male-Female-XLR-Connector-with-3-4-5-6-or-7-Pins-factory-xlr3-003b.html, which is `1`, `2`, `3`, and `G` for a connector with a discrete ground pin.

evanshultz, 2018-04-20:
@poeschlr @jkriege2 @Ratfink 
I would like to finalize pin names before I push my audio connector footprints. Right now the symbols for these footprints look like:
[image]

Does anybody have any issues with these pin names?

Any response to my comments on Jan 22 or the footprint name question above?

Ratfink, 2018-04-21:
The pin numbering seems fine to me.  As an aside, is it intentional that the switches in AudioJack3_Ground_Switch appear to be closed when the plug is inserted, but the others are opened when the plug is inserted; or is that an error in the symbol?

As for naming, I agree with Rene that they should follow the "Connectors for Specific Functions" convention.  The way you suggested for following that looks reasonable.

As for your other comments on 2018-01-22, the only thing I have to say is that I strongly suggest fixing the existing footprints in a separate PR from adding new footprints.  The PRs shouldn't conflict at all because of the .pretty format's one-footprint-per-file organization.

pushytoxin, 2018-05-05:
Hi @evanshultz
Do you have the footprint for the mono XLR combo jack? If not, I'm going to start working on it.

evanshultz, 2018-05-05:
@pushytoxin 
Yes, I've made footprints for all Neutrik XLR combo connectors. I have an open question about footprint names above and then I will start to upload them. I'll try to get on this early next week.

evanshultz, 2018-05-12:
@pushytoxin 
I didn't get this done last week and you must want the footprint now. Which Neutrik PN? I can at least save you some effort by giving my current file to you.

pushytoxin, 2018-05-14:
@evanshultz, that is awfully kind of you. If you have the NCJ5FI-H-0, and you uploaded it to your personal branch, I could check it out. Also, if you have the NMJ6HCD2, I'd compare it with my own as well.
But it's really not urgent, I still have some unrelated bugs with this board.

evanshultz, 2018-05-16:
@pushytoxin 
Yep! I've actually got all Neutrik XLR and phono jacks. I changed file extensions to TXT to upload.
[NCJ5FI-H.txt](https://github.com/KiCad/kicad-footprints/files/2011142/NCJ5FI-H.txt)
[NMJ6HCD2.txt](https://github.com/KiCad/kicad-footprints/files/2011145/NMJ6HCD2.txt)

Note that I was only working on fab and silk lines so far. Descriptions and 3D models aren't included. Pin names aren't aligned to the new mating symbols. And there's probably some polish that will be needed to get them merged. But it's a nice start that hopefully can hold you over until I get them done.

evanshultz, 2018-06-18:
Can I get any feedback on a couple of footprints? I'd like to make sure I'm not missing anything big before I move forward with updating all the rest of them and starting a PR.

[image]

A couple notes:
- These are all drawn in a "normal" orientation with XLR pin 3 at towards the bottom.
- I've drawn silk where the part touches the PCB and also where any elements hang above the PCB.
- Pin naming is as described above.
- Footprint naming is slightly modified from above because the series is in the MPN. (I find it a bit odd that we put the manufacturer name first for QFN, for example, but not here when in both cases the repo name is descriptive enough to make the Function/Standard/Type unhelpful and we could just jump straight into the manufacturer name.)
- Are the description and keywords suitable?
- A new repo, `Connector_XLR` would be added.

http://www.neutrik.com/en/xlr/aa-series/nc3faah
```
(module XLR_Neutrik_NC3FAAH_Horizontal (layer F.Cu) (tedit 5B282C41)
  (descr "3 pole female XLR receptacle, grounding: without ground/shell contact, horizontal PCB mount (http://www.neutrik.com/en/xlr/aa-series/nc3faah)")
  (tags "xlr neutrik")
  (fp_text reference REF** (at 3.81 8) (layer F.SilkS)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text value XLR_Neutrik_NC3FAAH_Horizontal (at 3.81 -1.7) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text user %R (at 3.81 -3.7) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_line (start -7.19 6.8) (end 14.81 6.8) (layer F.Fab) (width 0.1))
  (fp_line (start 14.81 -15.4) (end 14.81 -12.7) (layer F.Fab) (width 0.1))
  (fp_line (start -7.19 -15.4) (end 14.81 -15.4) (layer F.Fab) (width 0.1))
  (fp_line (start -7.19 -12.7) (end -7.19 -15.4) (layer F.Fab) (width 0.1))
  (fp_line (start -8.69 -8) (end -8.69 -12.7) (layer F.Fab) (width 0.1))
  (fp_line (start 16.31 -8) (end -8.69 -8) (layer F.Fab) (width 0.1))
  (fp_line (start 16.31 -12.7) (end 16.31 -8) (layer F.Fab) (width 0.1))
  (fp_line (start -8.69 -12.7) (end 16.31 -12.7) (layer F.Fab) (width 0.1))
  (fp_line (start 14.81 6.8) (end 14.81 -8) (layer F.Fab) (width 0.1))
  (fp_line (start -7.19 -8) (end -7.19 6.8) (layer F.Fab) (width 0.1))
  (fp_line (start -8.69 -12.7) (end 16.31 -12.7) (layer Dwgs.User) (width 0.1))
  (fp_line (start 16.43 -12.82) (end 16.43 -7.88) (layer F.SilkS) (width 0.12))
  (fp_line (start -8.81 -12.82) (end -8.81 -7.88) (layer F.SilkS) (width 0.12))
  (fp_line (start -7.31 -7.88) (end -7.31 6.92) (layer F.SilkS) (width 0.12))
  (fp_line (start 14.93 -7.88) (end 14.93 6.92) (layer F.SilkS) (width 0.12))
  (fp_line (start -7.31 -7.88) (end -8.81 -7.88) (layer F.SilkS) (width 0.12))
  (fp_line (start 16.43 -7.88) (end 14.93 -7.88) (layer F.SilkS) (width 0.12))
  (fp_line (start -7.31 -12.82) (end -7.31 -15.52) (layer F.SilkS) (width 0.12))
  (fp_line (start 14.93 -12.82) (end 14.93 -15.52) (layer F.SilkS) (width 0.12))
  (fp_line (start 14.93 -12.82) (end 16.43 -12.82) (layer F.SilkS) (width 0.12))
  (fp_line (start -8.81 -12.82) (end -7.31 -12.82) (layer F.SilkS) (width 0.12))
  (fp_line (start -7.31 -15.52) (end 14.93 -15.52) (layer F.SilkS) (width 0.12))
  (fp_line (start 16.81 -15.9) (end 16.81 7.3) (layer F.CrtYd) (width 0.05))
  (fp_line (start -9.19 -15.9) (end -9.19 7.3) (layer F.CrtYd) (width 0.05))
  (fp_line (start -9.19 7.3) (end 16.81 7.3) (layer F.CrtYd) (width 0.05))
  (fp_line (start -9.19 -15.9) (end 16.81 -15.9) (layer F.CrtYd) (width 0.05))
  (fp_line (start -7.31 6.92) (end 14.93 6.92) (layer F.SilkS) (width 0.12))
  (pad 1 thru_hole circle (at 0 0) (size 3.4 3.4) (drill 1.6) (layers *.Cu *.Mask))
  (pad 2 thru_hole circle (at 7.62 0.635) (size 3.4 3.4) (drill 1.6) (layers *.Cu *.Mask))
  (pad 3 thru_hole circle (at 3.81 4.45) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad "" np_thru_hole circle (at 0 -8.89) (size 1.6 1.6) (drill 1.6) (layers *.Cu *.Mask))
  (pad "" np_thru_hole circle (at 7.62 -3.81) (size 1.6 1.6) (drill 1.6) (layers *.Cu *.Mask))
  (model ${KISYS3DMOD}/Connector_XLR.3dshapes/XLR_Neutrik_NC3FAAH_Horizontal.wrl
    (at (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
)
```

http://www.neutrik.com/en/xlr/nc3mbv-sw
```
(module XLR_Neutrik_NC3MBV-SW_Switched_Vertical (layer F.Cu) (tedit 5B282ECC)
  (descr "3 pole male XLR receptacle, switching contacts, grounding: separate ground contact to mating connector shell and front panel, vertical PCB mount (http://www.neutrik.com/en/xlr/nc3mbv-sw)")
[... 116 lines cut ...]
  (fp_arc (start -6.15 0) (end -17.27 6.98) (angle 65.7) (layer F.SilkS) (width 0.12))
  (fp_arc (start -16.15 -11.5) (end -18.65 -9) (angle 180) (layer F.Fab) (width 0.1))
  (fp_arc (start 3.85 11.5) (end 6.35 9) (angle 143) (layer F.Fab) (width 0.1))
  (fp_arc (start -6.15 0) (end -11.61 -11.94) (angle 148.2) (layer F.SilkS) (width 0.12))
  (fp_arc (start -16.15 -11.5) (end -18.81 -9) (angle 180) (layer F.SilkS) (width 0.12))
  (fp_arc (start 3.85 11.5) (end 6.51 9) (angle 140.6) (layer F.SilkS) (width 0.12))
  (fp_circle (center -6.15 0) (end -17.27 6.98) (layer F.SilkS) (width 0.12))
  (pad 1 thru_hole circle (at 0 0) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad 2 thru_hole circle (at -12.3 0) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad 3 thru_hole circle (at -6.15 6.05) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad G thru_hole circle (at -6.15 10.45) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
[... cut ...]
  (pad TN thru_hole circle (at -9.15 -3) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad RN thru_hole circle (at -3.15 3) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad SN thru_hole circle (at -9.15 3) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (pad GN thru_hole circle (at -9.25 8.25) (size 3 3) (drill 1.2) (layers *.Cu *.Mask))
  (model ${KISYS3DMOD}/Connector_XLR.3dshapes/XLR-6.35mm_Neutrik_NCJ10FI-V_Vertical.wrl
    (at (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
)
```

evanshultz, 2018-06-26:
These could actually go into the existing `Connector_Audio`. Comment above edited.

Any feedback on these footprints?

evanshultz, 2018-09-04:
@poeschlr @Ratfink @antoniovazquezblanco (and anyone else too!)
Can I please get feedback on the footprints above? I'd like to get comments before I tidy up this large number of footprints and submit them. Thanks!

antoniovazquezblanco, 2018-09-05:
Being picky, according to F5.2:
```
3. Component value (footprint name) must be displayed on the F.Fab layer:
    b. Allowable text size = {0.1mm to 2.0mm}
    d. If there is enough space, it should be placed inside footprint boundaries
```
Given the lengthy names I would try to reduce font size so that it does not collide with the outline and try to avoid pads so that is readable without disabling that layer.

Again, I am being picky, don't take me to seriously 😉.

poeschlr, 2018-09-05:
Not a single footprint right now has the value inside the outline. So we either need to update all footprints or this rule.

evanshultz, 2018-09-05:
The quote above says `If there is enough space`, it doesn't say this must happen. Contrary to the above a few footprints do fit the value totally inside (like `Connector_Card:SD_TE_2041021`), but most are outside. And anyway, this field will be overwritten with the value pushed from the schematic during a PCB update so what is seen above may not be the actual test on the PCB anyway.

That all being said, I'll move the text below the footprint body and this is exactly the feedback I was hoping to get before I submitted a PR and then needed many updates. While I'm at it, probably I should scoot the silk ref des close to the top of the part body as well.

If there are any other comments I'll give it a couple days before starting the PR.

@antoniovazquezblanco 
Thanks for looking! Anything else?

evanshultz, 2018-10-04:
@DanSGiesbrecht 
Any comments on the above before I get started? Maybe a fresh set of detail-oriented eyes will see something that I can get a head start on?

DanSGiesbrecht, 2018-10-04:
@evanshultz unfortunately, I don't have the ability to take a look until this Saturday (when I'll have internet and time). I'll go over it when I can!

closed 2018-12-18 by evanshultz