KiCad/kicad-footprints #209: Most footprints for packages that have Exposed pads to not conform to KLC with regards to paste layer implementation

poeschlr, 2018-01-03:
Violated KLC rule: http://kicad-pcb.org/libraries/klc/F6.3/

This rule states that there should be one large copper pad (paste layer disabled. Mask enabled unless a specific mask size needs to be achieved.)
Overlayed with smaller pads for paste only. (This pads have no copper enabled and no pin number assigned.)

---

Currently a lot of footprints (nearly all of them) have multiple copper pads that have simply a negative paste clearance or solder paste ratio set.

krtkr, 2018-01-04:
Hey,
I can take a look at Package_QFP library on the issue.

poeschlr, 2018-01-31:
This has been fixed in footprints https://github.com/KiCad/kicad-footprints/pull/285 and https://github.com/KiCad/kicad-footprints/pull/329. (There might still be some small number of footprints remaining but the majority should now conform to the KLC.)

closed 2018-01-31 by poeschlr