KiCad/kicad-footprints #211: Should KLC specify the paste coverage ratio of the exposed pads?

poeschlr, 2018-01-03:
Currently we only state that the paste should be broken into multiple squares.
Should we specify the percentage of the area that should be covered?

Rule in question: http://kicad-pcb.org/libraries/klc/F6.3/

SchrodingersGat, 2018-01-06:
That's a tough one, this varies between manufacturers (based on their datasheet) and even between fab houses, depending on processes.

We could have a minimum / maximum coverage based on calculation of pad area? But this would require some much more complex checking...

poeschlr, 2018-01-09:
Has IPC something to say about that?

poeschlr, 2018-01-17:
I found this plog post that suggests the coverage should be between 50 and 80%
https://www.allaboutcircuits.com/technical-articles/stencil-design-for-exposed-pad-packages/

I think we should definitely mention that fact in the KLC. Until someone comes up wit a good way to check this, we will need to rely on manual checking of this.

closed 2018-01-31 by poeschlr