KiCad/kicad-footprints #214: Footprints with oval drill. Should we require alternative with circular drill?

poeschlr, 2018-01-03:
We have a few footprints that use oval drills.
Some of them also have a second footprint that uses a larger circular drill instead. Should we require such an alternative footprint (where possible) such that users who have not access to a fab that can create such a hole can use them?

SchrodingersGat, 2018-01-06:
I personally do not think so:

1. There is no guarantee that we can pick a radius that will actually "work" especially for long-thin oval drills
1. Pads can already be edited by the user if they need to
1. This adds an unnecessary burden to contribution, I think.

poeschlr, 2018-01-18:
Ok i think the decision here is: If somebody supplies an alternative footprint for one that has oval pads we will accept it, but it is not required. (We of course only accept it if the alternative drill size fits the footprint and if no other problems result from the circular drill.)

closed 2018-01-18 by poeschlr