KiCad/kicad-footprints #223: CP_Radial_SMD have fixed chamfer of 1mm

Shackmeister, 2018-01-04:
Hi
since the SMD radial capacitors have wrong models, I wanted to make sure the footprints are up to date before making the new models.
first thing I noticed is all the footprints has the same chamfer (polarity, not pin 1) of 1mm for all sizes.
should it be kept like this or wouldnt a percentage chamfer make more sense?
please pitch in some opinions so I can get started :)

poeschlr, 2018-01-04:
From KLC http://kicad-pcb.org/libraries/klc/F5.2/:

> Bevel should be 1mm or 25% of package size (whichever is smaller)

Shackmeister, 2018-01-04:
Ah I only thought it was for the pin 1 indicators.
Ill add the change it to min(width/4,1.0) for the footprints then. (and change the models accordingly)

poeschlr, 2018-01-05:
I had a closer look.
First of all i can't find the footprints you are referencing. (Are currently not in the lib under that name.)
Secondly: If it is not meant as a pin 1 marker but to represent the package outline then the package dimensions determine the size of the chamfer.

Shackmeister, 2018-01-06:
I guess the chamfers are meant as a pin 1 (polarity) indicator, but the are also a part of the packagedrawing.. Tricky part is the chamfer is not specified in the datasheet, so it would be a matter of checking manufacturer 3D models for the measurement.
we could go with the min(1.0,width/4) and call it worst case scenario, but that's not too accurate

poeschlr, 2018-01-06:
Would be good enough for now. If someone is motivated this can be done at a later date. And for generic footprints it will always be hard to find one   definition that fits all possible parts.

Shackmeister, 2018-01-07:
My plan is to grab all the different radial smd caps I have at work and see if they have a pattern (like 1/4 of the width) otherwise it will be 1mm,w/4 and models make accordingly :)