KiCad/kicad-footprints #224: Questions: Which pin is pin1 on a diode bridge rectifier?

jkriege2, 2018-01-05:
Hi!

I'm currently adding several diode bridges to the symbols, footpritns and 3D models. With the typical SIL-packages, I'm a bit stuck, as I'm not sure which pin to designate pin 1:
[image]
Could you quickly state your preference?

For now I took the left pin, but currently I tend towards using the pin on the right, marked by the package bevel! This would also prevent us from the situation that pin1 is marked on the silkscreen, but the package has a mark at pin 4 ...

Best,
JAN

SchrodingersGat, 2018-01-05:
Can you provide a datasheet for one of these packages? I assume that these are not simply the "standard" SIP package?

jkriege2, 2018-01-05:
e.g. https://diotec.com/tl_files/diotec/files/pdf/datasheets/b40c700040.pdf

jkriege2, 2018-01-05:
or http://www.vishay.com/docs/88657/kbu6.pdf

pointhi, 2018-01-05:
I would also tend to the body bevel on the right, because the datasheet doesn't specify a pin 1, and it's the visual noticable side where people normally assume pin 1 is located.

jkriege2, 2018-01-05:
OK! I think I'm going to rotate all packages as follows:
[2018-01-05 14_08_30-3d viewer](https://user-images.githubusercontent.com/11967523/34610557-fe1fc16e-f221-11e7-8b67-0c8a8f13d778.png)

SchrodingersGat, 2018-01-05:
Agreed. One of the examples had a longer pin ok the right side also. Definitely should be pin 1

jkriege2, 2018-01-06:
OK, so I tagged the marked pin (package bevel or longer pin) as pin 1, but in the footprint put them on the left to fulfil KLC (pin 1 top-left).

I will add some more of  these and post at some point in the  next week I think!

closed 2018-01-06 by jkriege2