KiCad/kicad-footprints #228: Connector_IEC_DIN

SchrodingersGat, 2018-01-06:
This legacy library has now been archived but the footprints still need to be transferred across to here..

poeschlr, 2018-01-09:
The generator scripts have not yet been merged. They are also not easily changed to fit new requirements. So i would suggest we move them over as they are right now and document remaining KLC problems in an issue. (One of the problems is the naming of the footprints.)

SchrodingersGat, 2018-01-09:
Are you able to take care of the transfer of these, Rene?

MRQ, 2018-01-10:
Are you referring to this generator script?: https://github.com/pointhi/kicad-footprint-generator/pull/101
You might have seen that I changed the naming. My rule is documented in the [generator script](https://github.com/pointhi/kicad-footprint-generator/pull/101/files#diff-090538bdac423c1b360263be38755ab2R14).

poeschlr, 2018-01-10:
Regarding naming, see http://kicad-pcb.org/libraries/klc/F3.6/ section `Connectors with Standardised Shape`

> Many connectors are provided by multiple manufacturers in an industry standard package or physical layout. These connectors should use the accepted industry naming at the beginning of the footprint name:
`[Standard]-[Pins]_[Series modifier]_[MAN]_[Series]_[MPN]_[Pin layout]_P[Pitch]_[Modifiers]_[Orientation]_[Options]`

In your case it should be at least:
`[Standard]-[Pins]_[Pin layout]_P[Pitch]_[Orientation]` (Pin layout only for connectors that have missing pins)

poeschlr, 2018-08-29:
The footprints in question have been merged.

closed 2018-08-29 by poeschlr