KiCad/kicad-footprints #235: FSDH321L and FSDL321L are missing an 8LSOP package

antoniovazquezblanco, 2018-01-09:
Datasheet: https://www.fairchildsemi.com/datasheets/FS/FSDH321.pdf

Components FSDH321L and FSDL321L available at the symbols repo (Converter_ACDC) are missing a footprint.

antoniovazquezblanco, 2018-01-09:
If at least we can agree on naming for the footprint here I can add it to the symbol in the current PR. Any suggestions @SchrodingersGat?

poeschlr, 2018-01-09:
Regarding naming see: http://kicad-pcb.org/libraries/klc/F3.4/ (Gull wing)

I think these might be possible as generic footprints so no need for a manufacturer prefix.
I assume they will be part of the SOIC lib and named `LSOP-8_ ...`

antoniovazquezblanco, 2018-01-10:
I guess then it would be `LSOP-8_6.4x9.7mm_P2.54mm`.

Thank you very much @poeschlr.

stephf0716, 2018-01-21:
I am working on the footprint. 
@antoniovazquezblanco 
Shouldn't name be `LSOP-8_6.4x9.2mm_P2.54mm`?
[image]

antoniovazquezblanco, 2018-01-22:
@stephf0716 you are right! Sorry! I will fix the footprint filter in the component. Thank you very much for your help ;)

stephf0716, 2018-01-22:
@antoniovazquezblanco I will use `Fairchild_8LSOP`

poeschlr, 2018-01-22:
`Fairchild_LSOP-8` would fit the rest of the library better.

PR #279 (2018-01-23): Fix #235: FSDH321L and FSDL321L are missing an 8LSOP package
Fix #235: FSDH321L and FSDL321L are missing an 8LSOP package
https://www.fairchildsemi.com/package-drawings/ML/MLSOP08A.pdf

[image]
[image]
commit 8268f3be (2018-01-23): Merge pull request #279 from stephf0716/add_lsop8

Fix #235: FSDH321L and FSDL321L are missing an 8LSOP package
closed 2018-01-23 by jkriege2