KiCad/kicad-footprints #238: Missing fuse holders

u382903, 2018-01-10:
Looks like good footprints for fuse holders were deleted. They were described as "Fuse_Holders_and_Fuses:Fuseholder5x20_horiz_open_universal_Type-III", type II and I (also other footprints with 5x20 and 5x25 in name were good). Please restore them.

poeschlr, 2018-01-10:
If you can link at least one product to each of them, then we can include them.

u382903, 2018-01-10:
Here you have list of many products that were compatible with these footprints - first 8 for example from link below:
https://www.digikey.com/products/en/circuit-protection/fuseholders/140
or here: https://www.tme.eu/gb/katalog/#id_category=100088&page=1&s_field=niski_prog&s_order=ASC
or this: https://www.tme.eu/gb/details/zhl77/pcb-fuseholders/stelvio-kontek/ptf77/
Type III was universal, it was made from type I and II.
So these footprints were good for 'bare' fuse clips, which are used widely.

stephf0716, 2018-01-20:
Please link to a specific product for each footprint. The first two links are not specific, and the suggested land pattern for the third link you provided does not match any of the three footprints you mentioned. 

Here are links to fuse clip datasheets from three manufacturers: [littelfuse](http://www.littelfuse.com/data/en/data_sheets/littelfuse_fuse-clips.pdf), [Keystone](http://keyelco.com/userAssets/file/M65-fuse-mountinglayout.pdf), and [Cooper](http://www.cooperindustries.com/content/dam/public/bussmann/Electrical/Resources/product-datasheets-a/Bus_Ele_DS_2131_1AXXXX_Series.pdf)

The recommended footprints for 5x20mm fuse clips do vary from part to part.

u382903, 2018-01-25:
For Type I this would fit: https://www.digikey.com/product-detail/en/littelfuse-inc/01000056Z/F3784-ND/2498892 (in catalog it's described as series 100)
For Type II: https://www.digikey.com/product-detail/en/littelfuse-inc/01110501Z/F4189-ND/2518598 (in catalog it's described as series 111)
And Type III would fit both of those fuse clips series.
Recommended footprints indeed vary from part to part, but I think there should be footprints for 'bare' fuse clips. It will be laborious process to add most of them, but these footprints will be good for beginning.

poeschlr, 2018-01-31:
I am not opposed to adding them as long as order information for at least one part is added to the description field of the footprints. That way users have a chance to find a part for the footprint. (you can add for example: "Generic footprint for fuse holders similar to: xxxxx, yyyyy, ..."

So could you open a pull request that transfers them over and also adds that stuff to the description field?

evanshultz, 2018-10-03:
I'll take this one.

In my experience 5x20mm and 6.3x32mm (1/4" x 1-1/4", 3AG, AGC) are the common cylindrical fuse sizes. Are there others anyone suggests we also include clips for while I'm at it?

I trolled the internet for the vendors above, vendors I knew and vendors at DigiKey to build up a small sampling of fuse clip info. Based on the parts I found (https://docs.google.com/spreadsheets/d/1OSzm-NI75B3XSdkQu7nbU0WIWMq-NZ3z9Eb7iX6Cmqw/edit?usp=sharing), there isn't broad commonality. I grouped rows in each configuration which are close enough to share a footprint using color. I choose "commonality" to mean <=0.5mm dimensional differences with ~1mm for drill diameter as I've seen both hand-built and mass production boards install the fuse in the clips before insertion to the PCB which means a bit of clip floating in oversized holes can be handled during wave solder.

Building off of http://kicad-pcb.org/libraries/klc/F3.7/, I propose the following footprint naming convention: `Fuseholder_Cylinder-[size x]x[size y]mm_Clip_***_[pin position]_Horizontal`. Open items:
1. `***` needs to capture the pin spacing, drill diameter, and potential body XxY size if we will have multiple footprints. However, that doesn't seem too helpful to the user trying to select a footprint. Maybe we instead make a universal footprint with several sets of overlapping holes so a single footprint can accommodate loads of clips?
1. These fuse clips come in `|` and `-` pin locations so `[pin position]` could be based on absolute position (`PinsVertical`, `PinsHorizontal`, `PinsBoth`) or relative to the fuse body (`PinsInline`/`PinsParallel`/`PinsLength`, `PinsNormal`/`PinsPerpendicular`/`PinsWidth`, `PinsBoth`). Since absolute could change if the footprint is rotated, I favor the latter style. Both seem rather unintuitive to me, though. Any better ideas?
1. Can `_Horizontal` be left off? I don't know any clips that are vertical, but I suppose there could be. Or the information is provides to the user is worthwhile.

For reference, here are the footprints that were lost when transferring footprints from the old library (https://github.com/KiCad/Fuse_Holders_and_Fuses.pretty):
[image]

feichtinger, 2018-12-28:
Is there any progress on this issue? As far as I can tell the footprints for 'clip-type' fuse holders are still missing.

poeschlr, 2018-12-28:
We laid out the requirements for these to be accepted above. Feel free to create a pull request complying with these requirements.

To be honest after reading the stuff even dug up i am more and more in favor of using part specific footprints here instead of going generic. (I know generic is tempting as it gives the impression of having lots of parts covered with a single footprint. However if the naming convention is already that complex then i have to ask the question if any user will then be able to find their correct footprint. There simply is too much variation between different parts. Especially if we take things like 3d models and part outlines into account as well. These will only add further parameters to be covered by a generic footprint naming scheme.)

evanshultz, 2018-12-28:
I already made a number of them and was waiting on more input to the above questions I raised. I'll push the footprints (but not until next week when I'm back at that computer) and then we can decide it if works or to scrap it.

herostrat, 2019-01-17:
I looked over the footprints and your research and firstly: awesome work.

The problem really is the sheer amount of different vendors and options.
I think that a "generic" version would be beneficial, because as far as I can see, versions and naming changes often and gets extended.

To your questions:
 1. imho a superposition of common spacings is a good idea. Because we have a multitude of very similar footprints, this could enable a short list of generic versions.
 2. because there is no "official" naming convention for this stuff, I think we should just refer them to "type-1" and so on. This makes the naming a lot more comprehensible
 3. I would not include horizontal

I see a lot of SMD fuseholder, but I do not know how common they are.
So I would probably indicate in the name if it is TH or SMD

My reasoning for those arguments is:
If we include generic footprints, the user has to verify that his/her fuse holder falls under this category be comparing the bought ones with the footprint of KiCad. 
In many cases (if we can make it right) the actual fuse will fall under the generic category.
If not the user has to modify the footprint and if he/she chooses to upstream it, we can think about extending the portfolio.

Another point I want to make:
On many cases the fuseholders get used for other stuff than just fuses (e.g. geiger tubes) or uncommon fuse sizes that are not sensible to include in a standard library (e.g. in my cheap multimeter they use a 5x10mm fuse).

Because most fuse holders are a single component, we could also add a wings as single footprints, enabling uncommon usecases.
Naming could be just Fuseholder_Universal_Type-X

evanshultz, 2019-01-23:
Sorry. It appears I made new fuse holder footprints and attached those above. I should have made a PR for them but had forgotten with the passage of time.

I agree with your points above, including adding "THT" to generic footprints, and thanks for reading and commenting. I do still favor a more intuitive name than "type x" because then on a BOM it would be more clear what type of fuseholder is being referred to (naturally a KiCad user would be able to easily see the footprint), but if everybody else like the "type x" style that's OK with me.

But then again, I only came up with 3 possible generic footprints. An alternative would be to have several exact footprints and then better specify them so the user could look at any datasheet and see if their desired fuseholder fit in our specific footprints. Does that proposal make sense? In this case, perhaps a naming convention something like `Fuseholder_Cylinder-[size x]x[size y]mm_[Man]_[MPN]_[pin position as described above]_P[pin spacing]x[outside spacing]mm_D[drill]mm_[Horizontal|Vertical]` would work?

Thoughts?

evanshultz, 2019-06-25:
@poeschlr 
@herostrat 
Any more comments on this? I would like to get the naming straightened out and submit a PR for these footprints, but I'd most prefer to settle the naming here so the PR is clean and discussion stays together.

poeschlr, 2019-06-26:
I like your last proposal more than the generic route to be honest. (As i fear that fuseholders are not standardized enough for investing time into a generic footprint. Especially one that would fulfill our standards of not only having the pads at the correct position but also the correct outlines on fab and silk as well as courtyard.)

evanshultz, 2019-07-18:
I submitted a PR, as you can see above, and pushed a number of footprints if anybody wants to take a look.

But I think there is some more good discussion we can have before merging anything.

---

Regarding the "type" of the fuse, I proposed something with intelligence above but realized the old footprints used "inline" and "lateral" already. Those terms are OK with me, so I adopted them. No reason to use two terms in the footprint name when "type 1" was the same as "inline" and the word "inline" also tells you something useful.

20.5 and 20.6mm are common outermost pin spacing for inline clips. The first link Google returned from querying "5x20mm fuse filetype:pdf" was http://www.farnell.com/datasheets/1504744.pdf, and that shows the metal end terminations are 5mm long on each side. So even having fuse clips 1mm offset from ideal (and maybe even more) should mechanically be OK for many fuse clips (some have a retention mechanism that could create interference) and likely has little to no impact on contact resistance and maximum carry current. So I think a generic footprint would be fine and useful.

The footprint naming I went for is `Fuseholder_Clip-[fuse size x]x[fuse size y]mm_[Man]_[MPN]_[Inline|Lateral|Universal]_[]mm_P[pin outermost spacing]x[pin spacing on clip]mm_D[drill x]x[drill y]mm_[Horizontal|Vertical]`. Please comment if you have some thoughts.

Lastly, it was mentioned above or somewhere else that a single footprint has two clips so the BOM doesn't quite match up with a single ref des. Even more than that, there is a fuse (so three total MPNs for one ref des). However, the ref des will be matched to one schematic symbol and there you want the fuse and it's electrical parameters selected. So the clips need to "ride along" as a mechanical part attached to the fuse (the electrical part) ref des. Same if it's a single box or tube-style fuseholder rather than clips. At least that's the way I've successfully handled it in other ECAD tools.

It would be great to have these footprints merged, so I'm happy to see more comments so all open questions can be answered and we can make some fuse clip footprints available. Thanks!