KiCad/kicad-footprints #240: KA5H0265RCTU and KA5H0265RCYDTU are missing a TO-220-5L package from fairchild

antoniovazquezblanco, 2018-01-10:
Datasheet: http://www.onsemi.com/pub/Collateral/KA5L0265R-D.PDF

The package seems to be particular for that manufacturer. The name is `TO-220-5L`.

I would suggest the names `Fairchild_TO-220-5L` or `ONSemi_TO-220-5L` (since Fairchild has been acquired by ONSemi). Which one should we use?

jkriege2, 2018-01-10:
Isn't this simply https://github.com/KiCad/kicad-footprints/blob/master/Package_TO_SOT_THT.pretty/TO-220-5_Vertical.kicad_mod ?
[image]

jkriege2, 2018-01-10:
oohhh ... just saw that the pin-spacing is not equal ... then we could simply modify the footprint above accordingly and add it under the name `ONSemi_TO-220-5L`. I would in addition add both companies to the description-field inside the kicad_mod-file

antoniovazquezblanco, 2018-01-10:
ONSemi it is then. Thank you very much @jkriege2!