KiCad/kicad-footprints #241: KA5H0265RCYDTU is missing a TO-220-5L(Forming) package from Fairchild/ONSemi

antoniovazquezblanco, 2018-01-10:
Datasheet: http://www.onsemi.com/pub/Collateral/KA5L0265R-D.PDF

The package seems to be particular for that manufacturer. The name is TO-220-5L(Forming).

I would suggest the names Fairchild_TO-220-5L_Staggered or ONSemi_TO-220-5L_Staggered (since Fairchild has been acquired by ONSemi). Which one should we use?

Related to https://github.com/KiCad/kicad-footprints/issues/240.

jkriege2, 2018-01-10:
see my comment there: https://github.com/KiCad/kicad-footprints/issues/240

antoniovazquezblanco, 2018-01-10:
Expected name is `ONSemi_TO-220-5L_Staggered` then. Thanks again @jkriege2!

jkriege2, 2018-01-10:
I would call it `ONSemi_TO-220-5L_Forming`, i.e. stick to the name set by the manufacturer (in the end this is a manufacturer-specific footprint)

antoniovazquezblanco, 2018-01-10:
I have found the `Staggered` word in similar footprints and that is why I decided to stick to it. See http://kicad-pcb.org/libraries/klc/F3.5/.

Are you sure it is better to use the `Forming` word?

antoniovazquezblanco, 2018-01-10:
Anyways, KLC specifies `StaggerOdd` and `StaggerEven` instead of `Staggered`.

jkriege2, 2018-01-10:
I know, we usually use Stagger, but for manufacturer specific parts, it makes sense to use the exact name the manufacturer uses for future easier reference ... see http://kicad-pcb.org/libraries/klc/F2.3/

antoniovazquezblanco, 2018-01-10:
Best name proposal until now:
`ONSemi_TO-220-5L_Forming`

Thank you!