KiCad/kicad-footprints #242: Should Fairchild_TO-220F-6L be renamed to ONSemi_TO-220F-6L

antoniovazquezblanco, 2018-01-10:
Given that ONSemi has bought Fairchild, should we rename `Fairchild_TO-220F-6L` to `ONSemi_TO-220F-6L`?

In that case, as pointed in https://github.com/KiCad/kicad-footprints/issues/240, it may be a good idea for the footprint description should mention both brands.

jkriege2, 2018-01-10:
If you rename (which is fine IMHO but not strictly necessary), you also need to add a PR to the symbols, which changes all footprint-field-entries and FPFilters linking to that one!

antoniovazquezblanco, 2018-01-10:
Yes, could be "scripted":
`sed -i 's/Fairchild\_TO-220F-6L/ONSemi\_TO-220F-6L/g' *.lib`
Beware! I have not tested it although it should not harm things too much...