KiCad/kicad-footprints #246: Mark Invalid on load OptoDevice.pretty

tochimoreno, 2018-01-11:
There is an error in the module Siemens_SFH900.kicad_mod
Line: 43
Offset 20

(pad 3 thru_hole roundrect (at 2.54 0) (size 1 1.6) (drill 0.7) (layers *.Cu *.Mask)(roundrect_rratio 0.25))
(pad 2 thru_hole roundrect (at 1.27 0) (size 1 1.6) (drill 0.7) (layers *.Cu *.Mask)(roundrect_rratio 0.25))

I change roundrect by circle and it works.

Version: 4.0.7-e2-6376~60~ubuntu17.10.1 release build

jkriege2, 2018-01-11:
Please note that this repo is meant for the new and upcoming v5 (or nightly-builds) ... for these this is not a bug, I think!

Can you confirm that, or do you use a recent nightly?

tochimoreno, 2018-01-11:
Ups, sorry.
No I'm using the v4 branch (4.0.7)

closed 2018-01-11 by jkriege2