KiCad/kicad-footprints #266: Footprints with complex pad shapes would benefit from the use of polygon pads.

poeschlr, 2018-01-17:
Polygon pads result in better paste and mask handling then overlapping pads.

The downsides are:
- not compatible with version 4. (But we already have some libs that include rounded rectangles. This already results in invalid libs.)
- currently not handled by stepup which makes checking footprint correctness a lot harder.

evanshultz, 2018-03-06:
If for no other reason than I would probably forget if I didn't write it down, here are a few footprints (or footprint families) that would benefit:
- SOT-89 at https://github.com/KiCad/kicad-footprints/tree/master/Package_TO_SOT_SMD.pretty
- https://github.com/KiCad/kicad-footprints/blob/master/Package_SON.pretty/TI_X2SON-4_1x1mm_P0.65mm.kicad_mod (this one also needs the prefix changed from TI_ to Texas_)
- Infineon MLPQs and PQFNs at https://github.com/KiCad/kicad-footprints/tree/master/Package_DFN_QFN.pretty

pointhi, 2018-07-01:
Lets start a list of libraries which use the old system and could profit from polygon pads:

* **Jumper.pretty** Fix: https://github.com/KiCad/kicad-footprints/pull/716
* **RF_Shielding.pretty**
* ... (please add other libraries you know)

evanshultz, 2018-07-02:
List started above.

evanshultz, 2018-09-27:
I've fixed all the outstanding items I could find above and DFN/QFN (that aren't scripted) and TO/SOT at https://github.com/KiCad/kicad-footprints/pull/981. Except for the one Wuerth SMD inductor noted in that PR where I haven't figured out how to make the footprint from FreeCAD. And Rene's PR that is modifying the RF shields.

Is there another footprint or topic that this issue should cover, or can this be closed when the aforementioned two PRs (mine + Wuerth inductor and Rene's) are merged?

evanshultz, 2019-12-12:
All of the list above is done, many of the #981 items have been merged (broken apart into separate PRs) and that includes the inductor.

closed 2019-12-12 by evanshultz