KiCad/kicad-footprints #283: Invalid libraries when adding

Resaj, 2018-01-21:
I've tried to add all the libraries to my project on Kicad, but some of them are declared invalid by Kicad. These are the libraries:
[image]

I've seen that TODO.md includes:
- "Audio Jacks" inside "Connectors", renamed as "Connector_Audio" outside "Connectors"
- "Barrel Jacks" inside "Connectors", renamed as "Connector_BarrelJack" outside "Connectors"
- "GS (SolderJumper)" inside "Connectors", renamed as "Jumper" outside "Connectors"
- "OptoDevices" renamed as "OptoDevice"
- "TO_SOT_Packages_THT" renamed as "Package_TO_SOT_THT"

I don't know what the problem is. For more information, my SO is Ubuntu 16.04 LTS and my Kicad version is 4.0.2. And this is my PCB library table, with the paths configuration:
[image]

poeschlr, 2018-01-21:
This repo is intended for kicad version 5 (or the nightly builds.)

For this reason we allow rounded rectangle and polygon pads.
We will make a compatibility release later on that allows (parts of) this libs to be used in kicad version 4.

In the meantime there is a script that details which footprints are to blame and optionally remove them. More details see:
https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269/2

Resaj, 2018-01-21:
Thanks for your fast reply! I'll keep it in mind.
If you consider it, you can close the issue.

bobc, 2018-01-23:
> This repo is intended for kicad version 5 (or the nightly builds.)

version 5 _ONLY_

That really needs to be in big letters on the front page. Currently, there is no indication anywhere apart from informal comments that the new repos are v5 _only_ and _not_ backward compatible to v4.

People in the know might realise that "for v5" means incompatible changes, but for the average user there is no reason for them to suspect that the "v5" repos are not backward compatible.

poeschlr, 2018-01-24:
I added notes to the readme files, the github.io download pages and to the repo descriptions.

I also made it more clear on the download page via the changes contained in PR https://github.com/KiCad/kicad-website/pull/264

evanshultz, 2018-02-06:
Can this be closed now?

Resaj, 2018-02-06:
I think yes. @poeschlr ?
I'll update Kicad to V5 when it is available for avoiding these problems.

closed 2018-02-06 by evanshultz