KiCad/kicad-footprints #284: QFN packages in other libs?

poeschlr, 2018-01-21:
While checking the stuff for adding the ep size to packages i found that there is a lib called Sensor_Current. This lib holds a QFN footprint.
I think this footprint should be in the dfn_qfn lib with all other qfn packages.

There might be other such footprints. (Don't have time to check this right now.)

stephf0716, 2018-01-21:
Yes, I see a few other footprints as well e.g. `Allegro_SIP-` that should be moved to their respective package libs. May I migrate footprints to their respective package libraries? I am also thinking it may make sense to combine to have just one Sensor.pretty for the leftover footprints. Thoughts?

stephf0716, 2018-01-22:
@evanshultz  ok got it, that makes sense. 

@poeschlr I think it's ok to leave such "one-off" packages as they are since there is a corresponding symbol lib.

poeschlr, 2018-01-25:
@stephf0716, @evanshultz if these are truly one of packages, should they also include the part number for which they are valid?

evanshultz, 2018-01-25:
They are a manufacturer-specific package, but not PN-specific. So I think it makes sense to leave them as is. This package can always be used in multiple MPNs and more could be coming in the future.

poeschlr, 2018-01-26:
If it is not mpn specific why is it in a different lib than the one where i would expect generic footprints?

evanshultz, 2018-01-27:
It is manufacturer-specific. And also specific to a particular range of parts from one manufacturer (their current sensor product line). The packages are not MPN-specific.