KiCad/kicad-footprints #290: Fix USB_A_Horizontal.kicad_mod

andrewcharnley, 2018-01-22:
Hi,

USB_A is footprint is obviously borked. I corrected it and have attached to this post (I don't have access to git right now to do a pull).

Cheers, Andrew

[](url)
[image]
[USB_A_Horizontal.kicad_mod.zip](https://github.com/KiCad/kicad-footprints/files/1651479/USB_A_Horizontal.kicad_mod.zip)

poeschlr, 2018-01-22:
Well you complain about the horizontal footprint and link a datasheet for a vertical part. 
The [naming convention for connectors](http://kicad-pcb.org/libraries/klc/F3.6/) has a [detailed image explaining this.](http://kicad-pcb.org/img/klc/connector_orientation.png)

There is no generic vertical usb connector in the current lib. To be honest i am not really sure we should even attempt to include generic footprints here. (I think the current generic symbols should include the manufacturer and part number information in their name.)

andrewcharnley, 2018-01-22:
Yes sorry, I've obviously forgot about a bit of work I did there.

So, USB_Vertical has the original issue (you can see the pins aren't spaced correctly). I've then copied this to make a USB_Horizontal at some point and then I'm complaining about the pins in it.

So USB_Horizontal here is correct as per the USB spec (which defines the layout) and can be used to correct USB_Vertical. It does however link to the incorrect 3dshape.

poeschlr, 2018-01-23:
The spacing might be standardized. But what about drill size, length of the oval mounting pad, height of the part (for 3d model), ...

evanshultz, 2018-01-29:
@andrewcharnley 
I feel the questions raised above are closed. Can this be closed? Are you able to submit a PR to improve the libraries?

Ratfink, 2018-04-07:
Definitely the right move.  For our future reference in case this topic comes up again, the USB standards specifically say that there is *not* a standard footprint.  From the USB 2.0 standard (emphasis mine):

### 6.9 PCB Reference Drawings
The drawings in Figure 6-12, Figure 6-13, and Figure 6-14 describe typical receptacle PCB interfaces. **These drawings are included for informational purposes only.**

From the USB 3.1 standard (emphasis mine):

### 5.3.1.2 USB 3.1 Standard-A Reference Footprints
**This specification does not define standard footprints.** Any footprint may be used as long as all mechanical and electrical requirements are met. Example footprints are provided for reference only.

From the USB Type-C 1.3 standard (emphasis mine):

### 3.2.1 Interface Definition
[...]
10. **This specification does not define standard footprints.** [...]