KiCad/kicad-footprints #296: Crystal/oscillator size and polarity marking

evanshultz, 2018-01-23:
Polarity markings aren't in accordance with IPC: https://github.com/KiCad/Crystals.pretty/issues/23 and https://github.com/KiCad/Oscillators.pretty/issues/6. Note that most (all?) are script-generated with more info at the first link.

https://github.com/KiCad/Crystals.pretty/issues/24 mentions a possible seems to point out a mismatch between a package name and the package size.

poeschlr, 2018-01-24:
This is a duplicate of https://github.com/KiCad/kicad-footprints/issues/156 and https://github.com/KiCad/kicad-footprints/issues/155. I assume it can be closed?

evanshultz, 2018-01-29:
I closed #156 because it didn't have any new info. But the other two have some overlap, but still some unique info. I'll add the above to #155 and close this one.

closed 2018-01-29 by evanshultz