KiCad/kicad-footprints #297: build/installation: the install rule is to generic and does not prevent reverse installation

tijuca, 2018-01-23:
While working on the Debian packaging for the footprint it happens to me that the install target in the generated Makefile is going into a loop and is installing all sources again and again.

So why this is happen?
For packaging Debian and Ubuntu is using a dedicated folder 'debian' for placing all the control files for the package build. The install target is simply using './' to declare the root for the installation. This includes also the folder 'debian' which I need to use. A simply solution is to add 'debian' to the exclude pattern. A better solution could be to only select folders matching '*.pretty'. For the former approach I have added a small patch.

[ignore-debian-folder-while-install-target-is-called.patch.txt](https://github.com/KiCad/kicad-footprints/files/1657629/ignore-debian-folder-while-install-target-is-called.patch.txt)

poeschlr, 2018-01-23:
@stambaughw i think you created the original make files for this repo. As i have very little experience with makefiles could you take a look at the proposed patch?

@tijuca on github one normally starts a pull request instead of sending a patch. That way the change and the discussion are connected correctly.

tijuca, 2018-01-24:
> @tijuca on github one normally starts a pull request instead of sending a patch. That way the change and the discussion are connected correctly.

Sorry, that is a bit overkill for me then. I don't want to clone every repository there I want to contribute a little fix. Only accepting PR via the GitHub platform can't be the right way to go.

poeschlr, 2018-01-24:
For the library it is the way it is.
You can even supply changes via the web interface. (No need for a local clone)

And you must have already cloned the repo. How else would you create a patch?

jsreynaud, 2018-01-24:
I had proposed a pull request to fix this: https://github.com/KiCad/kicad-footprints/pull/291
It only select .pretty directories...

tijuca, 2018-01-24:
@poeschlr 
For packaging I don't clone the repositor{y,ies} as I don't need taht to do, we have uscan which is downloading a tarball that is then imported in a local packaging git tree. Or we use some stupid scripting to get via GitHub API the source tarball.
For creating a patch I don't need any cloned repository at all, I'm still able to use the diff and patch commands to prepare classical patches. This patch was created by help of git-buildpackage as this is one of the helper tools that is used a lot from the Debian people that do packaging.

But if you think it's correct to push and force people to only use the "GitHub way" then I'm out at all and won't provide some bug fixing things in the future. There is more outside than the GitHub universe there I can spend my time. And I don't like this issue tracking system for stupid small things. It's mostly overhead for me as people seems impossible to run `git am foo.patch`.

Closing this report as @jsreynaud has pointed to the correct solution already.

closed 2018-01-24 by tijuca