KiCad/kicad-footprints #298: Parse error in Siemens_SFH900

leventelist, 2018-01-23:
When I read the module, I get this. This is with 4.0.7

Error loading netlist.
IO_ERROR: PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'/home/lev/kicad_library/kicad-footprints/OptoDevice.pretty/Siemens_SFH900.kicad_mod'
line 43
offset 20
from dsnlexer.cpp : Expecting() : line 369
from kicad_plugin.cpp : Load() : line 327

poeschlr, 2018-01-23:
Well this library is intended to be used in kicad 5 or in a nightly version that supports rounded rectangle and polygon pads.

This is already mentioned in both the download page and the readme file of the repository.

There exists a script that can remove incompatible footprints such that you can use the rest. More details see: https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269/2

---

There is a similar problem for symbol libs. There has been a file format update to version 2.4 from the version 2.3 that is understood by kicad 4.0.x. This again is incompatible. But even worse, kicad will not throw an error but simply not display contents of the dcm file.

We do not yet have a script to safely backport symbol libs. (The script would need to check if there are symbols in there that use new file format features before changing the library file header information if possible.)

poeschlr, 2018-01-23:
I will close this, as it is a duplicate of https://github.com/KiCad/kicad-footprints/issues/283

closed 2018-01-23 by poeschlr

leventelist, 2018-01-24:
Thanks for the information.