KiCad/kicad-footprints #299: Release the footprints

leventelist, 2018-01-23:
Is it intentional that the footprints repository has no tags that corresponds to any KiCad releases? So it is somewhat rolling release. It could be helpful to tag a specific commit for a specific KiCad release.

poeschlr, 2018-01-23:
This repo holds footprints for the v5 release. (See readme.md file) You might have noticed that kicad 5 has not yet been released.

poeschlr, 2018-01-23:
You might want to take a look at https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269 for a more detailed explanation.

Or on the library download page (a short note) http://kicad-pcb.org/libraries/download/

I will make this more clear on the download page.

stephf0716, 2018-01-23:
@poeschlr Perhaps this should be bolded/highlighted/all caps in the repo information and the README? Same for the other KiCad 5 lib repos.

poeschlr, 2018-01-23:
Ok i will try that. I will also make it more clear on the library download page over on the github library.

leventelist, 2018-01-23:
On Tue, 23 Jan 2018 23:35:04 +0000 (UTC)
Stephanie <notifications@github.com> wrote:

> @poeschlr Perhaps this should be bolded/highlighted/all caps in the
> repo information and the README? Same for the other KiCad 5 lib repos.
> 

Thanks, I missed that. Now this is clear.

Levente

poeschlr, 2018-02-02:
The information that these repos are for kicad 5 has been added everywhere where i could add it.

closed 2018-02-02 by poeschlr