KiCad/kicad-footprints #301: Connector_RJ/RJ12_E Problems

openelectron, 2018-01-24:
I cant seem to find a connector to fit Connector_RJ.pretty/RJ12_E.kicad_mod.

To me it looks like the mount pin holes are too small. Also, the pins seem to be in the wrong orientation as the current part could only be fitted to the opposite side of the board to the silkscreen.

Before I do a PR to fix this does anyone know why this part is this way or know a manufacturer part number for it?.

poeschlr, 2018-01-31:
As this connector does not include a datasheet link and is not named after a manufacturer part number it might be hard to find what part this is intended for. My suggestion would be to remove such generic parts and replace them with parts that include ordering information. (Naming should then follow the rule "Connectors for specific Functions" in the [Connector naming convention section of the KLC](http://kicad-pcb.org/libraries/klc/F3.6/))

In this case `RJ12_E_[Manufacturer]_[part number]`

mickaelv, 2018-03-07:
Hi @openelectron , Did you find a part matching with the RJ12e footprint ? 
I think that a wrong footprint too. Not the good orientation and fixing hole are not correct.

openelectron, 2018-03-07:
@mickaelv  I never found a part. I added a new connector in #347, however it still needs some work before it can be merged.

Pwuts, 2019-10-23:
@poeschlr The only RJ12 connector in the library is now the one created in #347, but a connector that looks a lot like RJ12_E can be found [all](https://nl.aliexpress.com/item/32767485682.html) [over](https://nl.aliexpress.com/item/32948988244.html) [AliExpress](https://nl.aliexpress.com/item/32826457723.html). It doesn't seem to have a clear brand or type (every reseller gives their own name as brand name) but it is widely available and can easily be found.

What would be your suggestion for a type name in this case? I have made a footprint to its specifications and would like to send in a pull request but I have no idea what to call it.