KiCad/kicad-footprints #309: Symbol refactor proposal

pointhi, 2018-01-27:
Proposal:

* remove all old footprints which do not use polygones (many of them are cleary drawn by hand and do not look professional at all)
* only specify height of a Symbol, not Width (because I generated them with consistent height)
* remove copper logos (useless duplication)

poeschlr, 2018-01-27:
Why remove copper logos? They might be useful for people who manufacture without silk screen. (I am one of them)

Could you give a list of the footprints that do not use polygons?

pointhi, 2018-02-02:
The Creative commons logos for example do not use polygons.

The thing with the copper logos is that they are simple duplicates with a different layer, and users could simply make them by editing the silk version a little bit. At least that was the discussion when I generated most of the logos in the past.

poeschlr, 2018-02-02:
If we follow that reasoning we don't need to include a lot of our footprints. Every handsolder version, thermal via version, ... is "just" a duplicate of another footprint.

I would keep them as they are useful to others. I don't see a good reason why they should be removed. (They do no harm)

pointhi, 2018-02-02:
I'm fine with that. Should the remaining logos also get copper duplicates? 

The current logo handling (due to kicad constrains) causes a lot of duplication in this directory at the moment.

poeschlr, 2018-02-02:
If somebody is willing to supply them then i think we should accept them.

It might also be a good idea to include the svg source files somewhere. assuming these footprints are generated using svg2mod. (allows for easier maintenance later on.)
Maybe adding a source directory within the .pretty folder that holds them could be the solution for that.

Ratfink, 2018-03-05:
I'm in favor of removing the bad-looking footprints in Symbol.pretty.  I guess we'd ideally replace them with better looking versions, but simply removing them would make the library look more professional.

I personally like that the width is specified as well as the height, and I'd like to see that expanded to all the remaining symbols.  For symbols like the KiCad logos and polarity symbols where the width is very different from the height, it can be helpful to know both dimensions before choosing which one to put on the board.  The ideal thing would of course be to have resizable symbols (in which case the size could be dropped from the name altogether), but I don't suppose that's immediately possible.