KiCad/kicad-footprints #317: Connector_card.pretty/SD_TE_2041021.kicad_mod problems

Resaj, 2018-01-29:
I've just actualyzed my repo downloading the original, but when I execute status command, git tells I have modified the file Connector_card.pretty/SD_TE_2041021.kicad_mod problems, although I haven't touch it.

Screenshot of terminal attached. The important information of the terminal is in English.

[image]

evanshultz, 2018-01-29:
This footprint got merged just a few hours ago. Could you pull the repo and try again just for sanity?

Also, could you clarify what "actualyzed" means in your post? I'm sorry but I don't follow. I suspect it's just tough to translate the concept you want to communicate into English.

Resaj, 2018-01-29:
Sorry for my limited English. Maybe if I write the commands I used you'll understand the problem.

I have done this process 3 times for actualyzed my local repo and my forked repo, and all times the result has been the same:
git fetch upstream                        // Upstream is the original repo of Kicad/kicad-footprint
git checkout master                     // I want to actualyze the branch master
git merge upstream/master         // Branch master is actualyzed
All is right now, I've done this process many times in several repositories.
But... here comes the problem, when I execute this command:
git status
You can see the result in the last image. The file has changed but anybody didn't modify it.
When I execute "diff" command, terminal shows the warning line about CRLF.

I've tried to delete my local repo and clone again, also I downloaded the conflicting file from my remote repo and from the kicad original repo and substitute it in my local repo, but any of this tries hasn't resolve the problem. Always the result is that the file has changed and the line endings are CRLF. So, I think the file has been commited with bad line endings.

poeschlr, 2018-01-30:
The repo does use the gitattribute text=auto ([see git docu](https://git-scm.com/docs/gitattributes#_checking_out_and_checking_in)) which should take care of line endings. But some versions of git have known problems with that on some operating systems.

In addition we also have a travis check to prevent such files from making it into the repo. ([It did complain but because it is only a warning i missed it.](https://travis-ci.org/KiCad/kicad-footprints/builds/334784326?utm_source=github_status&utm_medium=notification))

Fix committed: https://github.com/KiCad/kicad-footprints/pull/320

evanshultz, 2018-01-30:
@Resaj Fix merged. Please check now.

Resaj, 2018-01-30:
Solved! Thanks! 👍

closed 2018-01-30 by Resaj