KiCad/kicad-footprints #323: Connector_Harwin.pretty - pin numbering and silk orientation issues

andrewteasdale, 2018-01-30:
The new _Connector_Harwin.pretty_ library does not include some corrections to pin numbering and silkscreen orientation highlighted in a previously accepted pull request (see https://github.com/KiCad/Connectors_Harwin.pretty/pull/8). For example, _LTek-Male_2x04_P2.00mm_Vertical_ is as follows in the new library:

[image]

Whereas it should have silkscreen and pin numbering as follows:

[image]

Pin numbering/orientation for this family can be found on page 7 here https://cdn.harwin.com/pdfs/C005XX_M80_and_M83_Datamate_Series_Connectors.pdf.

The same pin numbering issue is also true for the Gecko connectors, as referenced on page 4 of https://cdn.harwin.com/pdfs/C125XX_G125_Gecko_Connectors.pdf.

Assuming there was no particular reason that these corrections were not included in the new library, I'm more than happy to submit another PR to fix this.

Incidentally, we use the Harwin LTek and M80/Mixtek connectors extensively so I'd be happy to take responsibility for extending the range of Harwin connectors available in this library.

poeschlr, 2018-01-30:
The original pull request that transferred them state they where rebuild from scripts https://github.com/KiCad/kicad-footprints/pull/89

The problem is i can't seem to find the scripts in https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts

I think i found them in olivers repo under https://github.com/SchrodingersGat/kicad-footprint-generator/tree/master/scripts/Harwin

oliver also has a branch called harwin-scripts. Not sure which version represents the current state of the footprints.

@SchrodingersGat could you give us a hint here?

SchrodingersGat, 2018-01-31:
The scripts on my repo should be the latest iteration (although I haven't touched them in a while). Feel free to make use of them (and improve as necessary).

Unfortunately I do not have time to help you guys with this. :)

poeschlr, 2018-01-31:
Is the one in your master branch or the one in the harwin branch more recent?

SchrodingersGat, 2018-01-31:
They appear to be the same (both 2 years old)

andrewteasdale, 2018-01-31:
Thanks - I'll get to work on making the required changes