KiCad/kicad-footprints #328: Naming convention in Converter_ACDC

antoniovazquezblanco, 2018-01-30:
Reviewing CI output in https://github.com/KiCad/kicad-symbols/pull/151 the following doubt arised for me:

Should all the footprints in Converter_ACDC be renamed to avoid containing the prefix `Converter_ACDC`?

poeschlr, 2018-01-30:
I would at least keep the prefixes ACDC and DCDC. As converter is already in the lib name i am open to drop that part of the names.

antoniovazquezblanco, 2018-01-31:
`ACDC` and `DCDC` is also in the library name. That is why the doubt arised. Seems a bit redundant to repeat the library name in the component name.

If you think the current convention is right I will leave it as is. Otherwise I would like to rename this before merging the referenced PR. 

I will do whatever you librarians consider best ;).

poeschlr, 2018-01-31:
That prefix is for making footprint filter definition easier.

antoniovazquezblanco, 2018-01-31:
Given that, I will leave it as is. Thanks

closed 2018-01-31 by antoniovazquezblanco

jkriege2, 2018-01-31:
@poeschlr: Yes that's true, but I think this approach only makes sense for components for which a generic symbol exists. the ACDC/DCDC-converter packages seem so specific to me that for most cases we will have specific symbols, so there is really no need for the prefix. I would vote for the simple MAN_PACKAGENAME-approach ...

antoniovazquezblanco, 2018-02-01:
I can open a PR for this (both symbols and footprints) if you reach consensus.

poeschlr, 2018-02-01:
I am ok with it. But this will mean that the converter libs are fully atomic from now on.

jkriege2, 2018-02-01:
Are there any generic ACDC/ACAC-converter symbols? I don't think so ... so I think that won't harm?