KiCad/kicad-footprints #334: Cleanup of Package_SOIC and Package_SSOP

jkriege2, 2018-02-01:
We have the two libraries
- https://github.com/KiCad/kicad-footprints/tree/master/Package_SOIC.pretty
- https://github.com/KiCad/kicad-footprints/tree/master/Package_SSOP.pretty
that assemble packages that "look" more or less the same. The only real distinguation is (as far as I can see) that the larger pin-pitches (1.27mm/2.54mm) are mostly in SOIC while the smaller ones are in SSOP (which matches the long-name forms "SOP"/"SO"=small outline package and xSSOP=Shrink Small-Outline Package), or they are distinguished by name ("if there is a second S it goes to SSOP"?), but no real rule. Also we have some "SSO"-packages in SOIC and some in SOPs in SSOP ...

Since this seems to be a real mess, I would like to clean that up. Only I'm not yet clear on how ... Proposals would be:
1. combine them both into "Package_SO_SOP_SSOP" (or something)
2. Find a clear rule, e.g. pin-pitch >=1.27 -> SOIC, pin-pitch <1.27 -> SSOP

What do you all think? @evanshultz @poeschlr

poeschlr, 2018-02-01:
I agree the current splitting is arbitrary and confusing. One question we must ask is: "Do we have time till the v5 release to fix this?"

The name Package_SO_SOP_SSOP is quiet unwieldy. And it does not contain mention of SOIC so it might confuse users.

If we combine them i would go for something like Package_SO, Package_SO_alike, Package_xSOxx, ...
Or maybe Package_GullWing (Might combine too many different footprints)

strombom, 2018-02-01:
When is the v5 release?

Can I do anything to help?

poeschlr, 2018-02-01:
Reviewing submissions by others always helps.

Helping with fixing issues is also a good way to help.

jkriege2, 2018-02-01:
I think I can do that transition this, or next week. Basically it should be simple: We simply combine all footprints into one Directory (I like Package_SO or maybe Package_SOIC_SSOP and run a search&replace on the symbols. 

Which name should we choose?

poeschlr, 2018-02-01:
Package_SOIC_SSOP has the benefit of connecting with the past. But long therm Package_SO might be better. For that reason i have a slight preference to Package_SO.

Add the change to the replace.json file. (So both Housing_SOIC and Housing_SSOP should point to the same lib.)
This is not to help your current search and replace but to have that file up to date.

strombom, 2018-02-01:
I made a pull request for the change.
https://github.com/KiCad/kicad-footprints/pull/335
https://github.com/KiCad/kicad-symbols/pull/269

Delete my pull request if you want to do it instead to ensure quality.

jkriege2, 2018-02-01:
Hi!

looks good so far. Four things are still missing (would be great if you could change those too):
1. As I noted in the footprint-PR: you need to update fp-lib-table
1. All the 3D model-pathes need to be corrected too
1. Finally we need the same PRs for the two 3D-model repos: https://github.com/KiCad/kicad-packages3D and https://github.com/KiCad/kicad-packages3D-source (3&4 combined ;-)

Best,
JAN

PS: Thanks for the effort!!!!!

strombom, 2018-02-01:
1. Updated: https://github.com/KiCad/kicad-footprints/pull/335/commits/8252156e2bb5e313670f268863a66833ed298985
2. Updated: https://github.com/KiCad/kicad-footprints/pull/335/commits/6dd463a1dfb9716647d20290250be8aae8c19ad7
3. Updated: https://github.com/KiCad/kicad-packages3D/pull/251

There were no SSOP or SOIC folders in 3D source, I assume they are all script generated?

jkriege2, 2018-02-01:
PRs look good to me ... @poeschlr @evanshultz Could you have a second look, then I (or you) can merge this ...

poeschlr, 2018-02-01:
I can look at it tomorrow. (If it has not been done until then.)

evanshultz, 2018-02-01:
@jkriege2 @poeschlr 
Yeah, these look OK to me too but I'll leave them to merge by one of you so I don't cut you off from any reviewing you may want to do:
- https://github.com/KiCad/kicad-symbols/pull/269
- https://github.com/KiCad/kicad-footprints/pull/335
- https://github.com/KiCad/kicad-packages3D/pull/251

jkriege2, 2018-02-02:
@poeschlr I'll wait for you to have a look, then I will merge these this evening ... or you can immediately merge them if you're happy!

OK?
JAN

Shackmeister, 2018-07-05:
@jkriege2 @evanshultz
Is this issue finished?

Ratfink, 2018-07-05:
All the PRs referenced here are merged, and there are no libraries called Package_SOIC or Package_SSOP anymore.  I'm confident enough that this is done that I'll go ahead and close it.

closed 2018-07-05 by Ratfink