KiCad/kicad-footprints #346: Copy KLC footprint chapter into this wiki

matyro, 2018-02-05:
It would be nice to add the Library convention section with the footprint guidelines into the project wiki. This makes it easier to find and faster to access.

https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention

poeschlr, 2018-02-05:
Why would this be faster to access? 
There is a direct link from this wiki page to the KLC section of the homepage. (In bold font)

Having two places for the same information adds a lot of maintenance effort, and an additional place for errors to sneak in. All of this for little to no gain.

matyro, 2018-02-05:
Ok to say the truth I never scrolled down until the links appear. I only clicked on the wiki button (which does not work by the way) and locked there for up to date information.

poeschlr, 2018-02-05:
Oh you mean we should make a wiki on the new repos. (you linked the old repos wiki so i was a bit confused how you could miss the bold notice about it being deprecated and the link to the new stuff.)

matyro, 2018-02-05:
Yes,
I talked about this repositorys wiki. Sorry for the confusion.

poeschlr, 2018-08-29:
A wiki just to link to the official kicad homepage is overkill. Added a link to the KLC in the readme this has to be enough.

closed 2018-08-29 by poeschlr