KiCad/kicad-footprints #360: Connectors contain non-metric drill holes

eldarkg, 2018-02-10:
Like 
0.65mm
0.81mm
1.016mm
1.99898mm
Should be 0.6 (or 0.7) mm, 0.8mm and 1mm and etc.
Affected libs (not full list):
Connector-IDC
Connector-Wire

poeschlr, 2018-02-10:
The only one that seems fishy is 1.99898
All others might be that way as suggested in the datasheet.

Also 0.65 is totally ok as normally fab houses have drills with differences of 0.05mm (or even finer)

eldarkg, 2018-02-10:
@poeschlr I don't think that such drill holes it is necessary in almost common cases.
This creates more drills in a board (like 1.016, 1.032, 1.042 and etc - shoud be one 1.1) - increase board cost.

My fab houses requires:
0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1.0, 1.1, 1.2, 1.3, 1.4, 1.5, 1.6, 1.7, 1.8, 2.0, 2.1, 2.2, 2.3, 2.4, 2.5, 2.6, 2.7, 2.8, 3.0 and additional from 3.1 to 4.5 with step 0.1, and 0.25, 0.35, 0.85 и 0.95 (mm).
And up to 7-9 different drills in one board.

eldarkg, 2018-02-10:
@poeschlr Why next??? The next should be the same.

Lib Name | FP Name | Pad Size | Drill Size 
--- | -------- | ------------ | -----------
Connector_IDC | IDC-Header_2x10_P2.54mm_Vertical | 1.7272 | 1.016
Connector_PinHeader_2.54mm | PinHeader_2x10_P2.54mm_Vertical | 1.7 | 1.0

eldarkg, 2018-02-10:
The KLC should reference to applicable drills set.

poeschlr, 2018-02-10:
Every user can change the footprints as they desire. The official lib will follow the datasheets.
And 0.05mm step between drill sizes is quite standard in most modern fabs. Most fabs will even round to the nearest available drill. (And now think about the pain for guys in the us. Nearly non of our drill sizes fit them.)

eldarkg, 2018-02-10:
@poeschlr I understand you. 
Are any other opinions?

poeschlr, 2018-02-10:
from your previous post:
> 1.016, 1.032, 1.042 and etc - shoud be one 1.1

Here i strongly disagree. We can not state that without looking at each part separatly. It might be that the current drill is already a bit large which could mean 1mm might be a better choice. Or 1.05 maybe even 0.95. (Will all depend on the specific part)

eldarkg, 2018-02-10:
@poeschlr I'm exaggerating. But in almost common cases this is true

evanshultz, 2018-02-10:
@eldarkg 
If you would be willing to compare datasheets against the footprints then we will perhaps find corrections to be made to the KiCad footprints. Otherwise, this case is speculation and subjective.

Our official library should match the datasheets exactly, unless there are extenuating circumstances. Consider the reverse: the official library deviates from the datasheet (and manufacturer recommendation) and there is a problem. That would obviously not be acceptable.

eldarkg, 2018-02-10:
@evanshultz In accordance [IDC Header datasheet](https://b2b.harting.com/files/download/PRD/PDF_TS/09185XXX324_100154466DRW008A.pdf)
Should be 1mm instead 1.016mm.

eldarkg, 2018-02-10:
@evanshultz In accordance [IDC Header with lock datasheet](https://b2b.harting.com/files/download/PRD/PDF_TS/09185XXX914_100154464DRW023A.pdf) should be 1mm instead 0.91mm.

eldarkg, 2018-02-10:
@evanshultz In accordance [IDC Header with lock horizontal](https://b2b.harting.com/files/download/PRD/PDF_TS/09185XXX903_100154464DRW016A.pdf) datasheet should be 1mm instead 0.81mm.

eldarkg, 2018-02-10:
@evanshultz SolderWirePad_3x_2mmDrill: What datasheet say you to use 1.99898mm for wire pads?!

eldarkg, 2018-02-10:
@evanshultz In accordance [PinHeader 1.27mm](https://www.mouser.com/ds/2/181/M50-353-1100851.pdf) should be 0.6mm instead 0.65mm.

poeschlr, 2018-02-10:
The solder wire says 2mm in the name. I would guess the 1.99898 is some sort of rounding error from back when the file was converted to the new mm based file format.

@Shackmeister i think you made the pin header footprints. Do you remember where you got the drill size from? (There is no datasheet in the description field)

Shackmeister, 2018-02-11:
Pretty sure I used the ones already in the script. Ill check up on it tomorrow, when im at my desk