KiCad/kicad-footprints #363: Resistor metric names

strombom, 2018-02-10:
The 2512 resistor footprint is called R_2512_6332Metric

Shouldn't the metric part be 6432?

poeschlr, 2018-02-10:
The metric names are sometimes hard to determine.
The 2512 is part of the table found in http://kicad-pcb.org/libraries/klc/F3.1/ This table is copied from wikipedia.

strombom, 2018-02-10:
Maybe wikipedia is wrong?

[image]

[image]

[image]

poeschlr, 2018-02-10:
I did not question this list but i looked at many other sizes and found there seem to be different conventions used. If you think this one should be 6432 then you can change it. (As other imperial codes with 25 in them use 64 as the metric equivalent, this might really be the case.)

The footprint generator is found in https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/SMD_chip_package_rlc-etc.
But you will also need to change the table on the website and the 3d models.

strombom, 2018-02-10:
Ok, looking at the script, they are referencing an IPC document that actually say that the metric name is 6332. So I guess it's Digikey, Rohm, Panasonic and Yageo that are wrong, a little bit frustrating :)

closed 2018-02-21 by strombom

KarlZeilhofer, 2020-07-16:
I stumbled on this issue too now. 

Farnell finds 9000 parts with 6432 and zero parts with 6332. 
It makes more sense, due to the 2:1 apsect ratio. 

Please reopen this issue.