KiCad/kicad-footprints #365: DirectFET footprints and pinout

evanshultz, 2018-02-11:
@Shackmeister 
At https://github.com/KiCad/Package_DirectFET.pretty/pull/4 you mentioned a script for the DirectFET packages. It seems these packages were ported faithfully to https://github.com/KiCad/kicad-footprints/tree/master/Package_DirectFET.pretty. Did you upload the script to https://github.com/pointhi/kicad-footprint-generator? I'm not seeing it there.

There are a number of missing footprints, so having the script would allow more to be easily added. And also see how our existing footprints are built.

More importantly, how did you determine the pinout? In order of pins 1, 2, and 3 it appears you chose D, G, S (hopefully I got that right). I want to create the symbols for DirectFET transistors but I'd like to know if the footprint pinouts are consistent. To me, using 1, 2, and 3 as G, D, and S makes the most sense (through all permutations are being used in Transistor_FET.lib already...). And since there are no symbols for these footprints I can see it may not be too late to sort this out.

Also, the ME footprint has 2 G pins and 4 S pins, while it should be 5 S and 1 G.

Shackmeister, 2018-02-11:
@evanshultz only the 3D models are scripted,  The footprints are handmade. The numbering used was from the original footprints (which had a number and a letter version) I have no idea what number scheme is best to use, so feel free to change it however you (and René) want

poeschlr, 2018-02-11:
I have no preference for pin numbering here. If the datasheet does not specify anything you are free to use any sensible numbering scheme (Including using letters instead of numbers.)

evanshultz, 2018-02-12:
@Shackmeister 
I just submitted https://github.com/KiCad/kicad-symbols/pull/297, which adds all the current non-automotive symbols for Infineon transistors using DirectFET packages. Below are the missing footprints that I'll be making. If you're interested in doing more modeling, you could add them to your script.
- L6
- MC
- MD
- MF
- S3C
- SA (note that this one is for _two_ transistors with a common drain)
- SB

@poeschlr 
For simplicity, I used the existing pinout of the footprints for the symbols.

evanshultz, 2018-02-13:
@Shackmeister 
Have you uploaded your 3D model script somewhere? I'm wondering if I can build off of it to make a DirectFET footprint generator. Otherwise, I'll make my own footprint generator script.

Shackmeister, 2018-02-13:
Hi @evanshultz
3D script is here:
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Directfet_SMD
But I think it's easier to make the footprint script from scratch with Pointhi's framework. Perhaps the scripts can share a yaml file for parameters