KiCad/kicad-footprints #374: Add missing footprints for STM32 microcontrollers

Ratfink, 2018-02-15:
Once https://github.com/KiCad/kicad-symbols/pull/305 and https://github.com/KiCad/kicad-library-utils/pull/212 are merged, there will be quite a few missing footprints needed for the STM32 microcontroller family.  Below is a list of the needed footprints, with links to datasheets that contain package/footprint drawings.  Naming for WLCSP footprints follows the suggestion [here](https://github.com/KiCad/Housings_CSP.pretty/issues/8#issuecomment-325613650), but without the ball pitch because the packages are already uniquely identified without it.

- [ ] Package_BGA:LFBGA-100_10x10mm_Layout10x10_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00161566.pdf
- [ ] Package_BGA:LFBGA-144_10x10mm_Layout12x12_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00191185.pdf
- [ ] Package_BGA:TFBGA-64_5x5mm_Layout8x8_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00251732.pdf
- [ ] Package_BGA:TFBGA-100_8x8mm_Layout10x10_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00166116.pdf
- [ ] Package_BGA:TFBGA-216_16x16mm_Layout15x15_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00071990.pdf
- [ ] Package_BGA:TFBGA-265_14x14mm_Layout17x17_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00387108.pdf
- [ ] Package_BGA:UFBGA-64_5x5mm_Layout8x8_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00039193.pdf
- [ ] Package_BGA:UFBGA-100_7x7mm_Layout12x12_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00098745.pdf
- [ ] Package_BGA:UFBGA-132_7x7mm_Layout12x12_P0.5mm (exists, but should be renamed with Layout): http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00111457.pdf
- [ ] Package_BGA:UFBGA-144_7x7mm_Layout12x12_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00141306.pdf
- [ ] Package_BGA:UFBGA-144_10x10mm_Layout12x12_P0.8mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00213872.pdf
- [ ] Package_BGA:UFBGA-169_7x7mm_Layout13x13_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00071990.pdf
- [ ] Package_BGA:UFBGA-201_10x10mm_Layout15x15_P0.65mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00237391.pdf
- [ ] Package_DFN_QFN:QFN-28_4x4mm_P0.5mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00104043.pdf
- [ ] Package_DFN_QFN:QFN-48-1EP_7x7mm_P0.5mm_EP5.6x5.6mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00105814.pdf
- [ ] Package_DFN_QFN:QFN-36-1EP_6x6mm_P0.5mm_EP4.1x4.1mm: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00210837.pdf
- [ ] Package_CSP:ST_WLCSP-25_Die425: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00140359.pdf
- [ ] Package_CSP:ST_WLCSP-25_Die444: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00104043.pdf
- [ ] Package_CSP:ST_WLCSP-25_Die457: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00206508.pdf
- [ ] Package_CSP:ST_WLCSP-36_Die417: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00108217.pdf
- [ ] Package_CSP:ST_WLCSP-36_Die440: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00039193.pdf
- [ ] Package_CSP:ST_WLCSP-36_Die445: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00105814.pdf
- [ ] Package_CSP:ST_WLCSP-36_Die458: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00214043.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die423: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00086815.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die431: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00115249.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die433: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00102166.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die435: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00257211.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die438: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00092070.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die439: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00093332.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die447: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00141133.pdf
- [ ] Package_CSP:ST_WLCSP-49_Die448: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00098745.pdf
- [ ] Package_CSP:ST_WLCSP-63_Die427: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00048356.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die414: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00191185.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die427: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00048356.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die435: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00257211.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die436: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00034689.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die441: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00213872.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die442: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00115237.pdf
- [ ] Package_CSP:ST_WLCSP-64_Die462: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00340475.pdf
- [ ] Package_CSP:ST_WLCSP-66_Die411: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/CD00237391.pdf
- [ ] Package_CSP:ST_WLCSP-66_Die432: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00101621.pdf
- [ ] Package_CSP:ST_WLCSP-72_Die415: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00108832.pdf
- [ ] Package_CSP:ST_WLCSP-81_Die415: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00108832.pdf
- [ ] Package_CSP:ST_WLCSP-81_Die421: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00141306.pdf
- [ ] Package_CSP:ST_WLCSP-81_Die463: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00282249.pdf
- [ ] Package_CSP:ST_WLCSP-90_Die413: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00037051.pdf
- [ ] Package_CSP:ST_WLCSP-100_Die422: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00094064.pdf
- [ ] Package_CSP:ST_WLCSP-100_Die446: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00118585.pdf
- [ ] Package_CSP:ST_WLCSP-100_Die452: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00330506.pdf
- [ ] Package_CSP:ST_WLCSP-100_Die461: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00284211.pdf
- [ ] Package_CSP:ST_WLCSP-104_Die437: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00034689.pdf
- [ ] Package_CSP:ST_WLCSP-143_Die419: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00071990.pdf
- [ ] Package_CSP:ST_WLCSP-143_Die449: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00166116.pdf
- [ ] Package_CSP:ST_WLCSP-144_Die470: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00366448.pdf
- [ ] Package_CSP:ST_WLCSP-168_Die434: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00219980.pdf
- [ ] Package_CSP:ST_WLCSP-180_Die451: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00273119.pdf

poeschlr, 2018-04-25:
I assume the TFBGA-216_16x16mm_Layout15x15_P0.8mm: should be 13x13mm. If so you can tick it off. otherwise the 16x16mm version is still missing

Ratfink, 2018-04-25:
I just checked all the relevant datasheets, and yes, that is indeed an error.  Thanks for catching it!  I'll make a PR to fix the symbols momentarily.

poeschlr, 2018-04-26:
So looks like we got all of them. Which libs in https://github.com/KiCad/kicad-symbols/issues/520 should now be fixed (if any)?

Ratfink, 2018-04-26:
With this done, all the MCU_ST_STM32* libraries have correct footprint associations.  Thanks for reviewing all of the footprints!

closed 2018-04-26 by Ratfink