KiCad/kicad-footprints #377: Asymmetric varistor footprints

aimylios, 2018-02-15:
All of the varistor footprints have been converted to be "polarized" (i.e. Pin 1 has a rectangular instead of a circular pad) as part of PR #48 (see commit 2e41670).

I do not understand why this change was made. Although [KLC F7.3](http://kicad-pcb.org/libraries/klc/F7.3/) states that "Through hole components should set the shape of Pin 1 to Rectangular" it also includes an exception: "Non polarized parts (such as THT resistors) should not set the shape of Pin 1 to Rectangular. For these (non polarized) footprints, all pad shapes should be consistent."

Varistors are non-polarized devices, and their package is (usually) fully symmetric. Unless there is a valid reason to have asymmetric footprints I would request to revert commit 2e41670. If the maintainers agree I can also submit a pull request to do this.

Ratfink, 2018-03-05:
I agree, varistors are not polarized devices, so making pin 1 rectangular doesn't make sense.  I'm in favor of reverting the commit in question.

evanshultz, 2018-03-05:
The existing footprints were moved from the old repo to the new one in the PR above, among other things. Changing to a rect pin 1 was not the only thing done in that commit, and the commit should not be reverted (because the commit was NOT only changing pin 1's shape).

@aimylios 
Thanks for catching this! You are correct that varistors are not polarized and I'd welcome a new PR that corrects the footprints to have all circular pads. Are you willing to do that?

Ratfink, 2018-03-05:
How do you mean, it wasn't only changing pin 1's shape?  I looked through the diff for commit  2e41670373493a62d1f8f3876b11526354db669e, and I see nothing other than changing "circle" to "rect" for pad 1 of every footprint in Varistor.pretty.  PR #48 as a whole did a lot more than that, but we're only talking about undoing one commit, not the whole PR. 🙂

evanshultz, 2018-03-05:
Oh. Yes. My mistake. I'll blame it on reviewing too many PRs lately.

@Ratfink You want to revert with your newfound powers?

Ratfink, 2018-03-05:
Well, git can't figure out how to revert the commit, so I've done it slightly less automatically and submitted it as #411.

aimylios, 2018-03-05:
Thanks for all of your comments! I didn't expect so much activity after more than two weeks of silence.
And thank you @Ratfink for taking care of this. PR #411 looks good, please merge!

PR #411 (2018-03-05): Use circular pads for varistor footprints
Commit 2e41670373493a62d1f8f3876b11526354db669e mistakenly changed the varistor symbols to use square pads for pin 1.  This commit reverts that change, as varistors are non-polarized devices.

Fixes #377.

------------

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the footprint(s) you are contributing
- [ ] An example screenshot image is very helpful 
- [ ] If there are matching symbol or 3D model pull requests, prov [... cut]
commit 9fc4b01e (2018-03-05): Use circular pads for varistor footprints

Commit 2e41670373493a62d1f8f3876b11526354db669e mistakenly changed the
varistor symbols to use square pads for pin 1.  This commit reverts that
change, as varistors are non-polarized devices.

Fixes #377.
closed 2018-03-05 by evanshultz